OpenEVSE/openevse_esp32_firmware #771: Source code provide in the relaese V5.0.2 web page is without openevse-gui-v2 associated code

zerodur12, 2023-11-26:
I try to re compile by my self the relaese V5.0.2 using Source code (zip) provide by @jeremypoulter in https://github.com/OpenEVSE/openevse_esp32_firmware/releases/tag/v5.0.2
I have notice that some parts of the source code in the zip file are missing:  the openevse-gui-v2 associated code needed to re-compile is missing.
I have download the missing file from https://github.com/OpenEVSE/openevse-gui-v2/tree/9fde4b2b213d62c94513c32edc488e9bcec3629d
but with no insurance that it was the file use when compiling the relaese V5.0.2
For the next relaese is it possible to provide a  "self content" source code file?

jeremypoulter, 2023-11-26:
This is a GitHub thing, we do not package the Zip file, and it looks like the Zip files do not contain sub modules :(

zerodur12, 2023-11-26:
Thanks for the answers. In that case why there isn't the creation of release for the openevse-gui-v2?

jeremypoulter, 2023-11-27:
The Git hash of the GUI is tagged as part of the submodule in the top level Git repository, so if you do a checkout of the firmware repository (and associated submodules) you are guaranteed to get the same source as used to build the released firmware.

Also building the firmware does not actually need the GUI source. The results of building the GUI are checked in to the firmware repository and we have checks in place to make sure these are consistent with what would be built from the GUI.