OpenEVSE/openevse_esp32_firmware #773: Simulating a Connection

1madison24, 2023-12-05:
Hello,

I am attempting to simulate a connection that allows me to set limits and charge rates (Basically activate the status bar for use) with out physically mounting to my open evse system. Obviously I am not trying to physically charge anything but I would like to do this simply for testing load management. I have changed the power status tile to max power and pointed it to energy but cannot see any changes since I'm not physically mounted. Any tips?

Thank you for your hard work

[image]
[image]
[image]

1madison24, 2023-12-05:
Also not sure if this makes a difference in the answer but I am using the esp32 wrover kit so that is the only hardware I want to be connected to for my simulation

glynhudson, 2023-12-15:
Are you looking for a hardware or software solution? Here's a hardware solution: https://shop.openenergymonitor.com/ev-simulator-kit/

jeremypoulter, 2024-01-08:
That is largely what I use for development, an OpenEVSE, a EV simulator and whatever ESP board I am using, although I have resently discovered [Wokwi](https://wokwi.com/) and the OpenEVSE firmware runs on that to some degree. I need to do some more experiments on it. I would also like to look at EpoxyDuino as a solution to build the OpenEVSE firmware for Linux. Both these options should allow using simulating the OpenEVSE on the dev machine.

matth-x, 2024-01-26:
@jeremypoulter Just would like to comment on the Linux support. I'm a big fan of that idea! A strength of OpenEVSE is that it's easy to add support for new charging management systems. With Linux support, that would even require no hardware at all.

I'm also a big fan of Mongoose and think that their hardware abstraction layer would suit OpenEVSE well (most notably the filesystem and millis function). A HAL, including existing Mongoose features, could encapsulate all platform-specific code or replace it with stubs. But that would require big changes in the code (much more than emulating the ESP) and just should be an inspiration for how to approach multi-platform support in future.

Anyway, if you can imagine going that direction, I would offer to take over some of the necessary tasks. We can also discuss this anytime in future too.

jeremypoulter, 2024-01-26:
The EpoxyDuino library I have been using for divert_sim looks to work well, I haven't tried any of the network/WiFi side but Mongoose at least should cover the networking so it should be possible to do a Linux 'port' I think. Probably should try and move to Mongoose 7

closed 2026-09-21 by github-actions[bot]