OpenEVSE/openevse_esp32_firmware #781: Matter Support

uppercaseVar, 2024-01-01:
Would it be possible to add matter support so that the EVSE can be controlled by a smart home app like Google Home, Alexa, HomeKit, or Samsung SmartThings? I know matter doesn’t officially support an EVSE device type but until it does the outlet device type can be used.

chris1howell, 2024-01-05:
Looks like it is possible. There is a ESP32 library for Matter.

https://github.com/espressif/esp-matter

ausil, 2024-05-09:
matter 1.3 was just released which introduced support for EVSE's https://csa-iot.org/newsroom/matter-1-3-specification-released/ It would be great to add support to simplify integration into smart home ecosystems

KipK, 2024-05-10:
Esp32 matter needs a bunch of ram based on what I saw. Plus squetch size would be too big on previous evse wifi module. Matter needs BLE compiled in because of Bluetooth commissioning.
The new wifi module with bigger flash, + some SPRAM could make it. 

An interesting discussion about it here: https://github.com/espressif/esp-matter/issues/363

quicklywilliam, 2024-10-11:
Curious if this is possible on the current WiFi module. The recent catastrophic demise of Juicebox/EnelX has me thinking a lot about the importance of chargers that support widely adopted open standards. For commercial chargers that’s OCPP, for residential it seems like Matter is probably the best bet?

RAR, 2026-09-21:
Some measured numbers, since the picture has changed since 2024.

**The stack is now in the toolchain.** pioarduino / Arduino-ESP32 3.3.x ships a prebuilt `esp_matter` plus a first-party `Matter` Arduino library, so no IDF-component build is needed on the core-3 envs. Commissioning without BLE is a supported path (`MatterOnNetworkWiFi`: device already on WiFi, pairing code + QR shown on-network) — which is the right one here anyway, since an OpenEVSE is always on WiFi/Ethernet before anyone would pair it, and classic ESP32 has no CHIPoBLE in the prebuilt libs at all.

**Footprint on this firmware's full stack**, one `MatterOnOffPlugin` endpoint, `-Os`, measured on an S3 build (where the link has room) against a true baseline:

| Section | Baseline | + Matter | Delta |
|---|---|---|---|
| Flash (text + rodata) | 1.86 MB | 2.55 MB | **+682 KB** |
| IRAM | 81 KB | 104 KB | +23 KB |
| Static DRAM (.data + .bss) | 71.6 KB | 133.8 KB | **+62 KB** |

Runtime heap not measured yet; Espressif's own figure for the CHIP stack is ~100 KB, with the data model placed in PSRAM where present.

**What that means per board:**
- **4 MB modules (openevse_wifi_v1 and friends)** — out. The app partition is already at ~98 %.
- **16 MB WROOM without PSRAM (the TFT board)** — the link overflows DRAM by ~10 KB before heap is considered, and those units run at `heap_largest` 20–26 KB with MQTT-TLS, tsdb and the display up. Out.
- **PSRAM boards (ESP32-S3, ESP32-P4)** — fits: 40.8 % of a 6.5 MB OTA slot, 134/327 KB static. Viable.

So this is a PSRAM-board, opt-in feature (`ENABLE_MATTER`, like OCPP), not something the shipped 4 MB module can carry.

**Device type.** esp-matter supports the Matter 1.3/1.4 EVSE device type (Energy EVSE + Device Energy Management + Electrical Sensor), but the Arduino wrapper has no EVSE endpoint yet and hub support for the EVSE type is uneven (Home Assistant handles it; the others need checking per hub). The pragmatic order is an On/Off Plugin Unit mapped to an `EvseManager` claim (exactly how the HA/MQTT/evcc clients already control charging) plus Electrical Sensor for power/energy, then the Energy EVSE cluster.

**Open question that needs a board, not a compiler:** CHIP is built against the IDF `mdns` component — the same instance the firmware's `ESPmDNS` uses, and Espressif documents `MDNS.begin()/end()` as breaking Matter discovery. #1271's move to a Mongoose-owned mDNS responder is a step in the right direction for this.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

RAR, 2026-09-21:
I'll look at getting it in a future S3 build, but it would require a hardware upgrade to run it.

RAR, 2026-09-21:
Follow-up: built on the actual target env rather than the prebuilt Arduino libs. `openevse_s3_lcd` (Arduino as IDF component, ESP32-S3 + 8 MB PSRAM) with `espressif/esp_matter 1.5.0` as a managed component, CHIP and the data model steered to PSRAM (`CONFIG_CHIP_MEM_ALLOC_MODE_EXTERNAL`, `CONFIG_ESP_MATTER_MEM_ALLOC_MODE_EXTERNAL`), Bluetooth off, one On/Off Plugin Unit endpoint bound to an `EvseManager` claim and `Matter.begin()` called once the network is up.

| Section | `openevse_s3_lcd` | + Matter | Delta |
|---|---|---|---|
| Flash | 2.62 MB (40.0 %) | 3.11 MB (47.4 %) | **+486 KB** |
| Static DRAM (.data + .bss) | 73.9 KB | 134.1 KB | **+60 KB** |
| IRAM | 82.1 KB | 83.4 KB | +1.2 KB |

Smaller than the prebuilt-libs figure above because there's no NimBLE/CHIPoBLE in this configuration. The +60 KB static is CHIP's own singletons (`chip::Server`, the interaction-model engine, …), which PSRAM cannot take; everything CHIP allocates at runtime goes to PSRAM under this config.

**These are link-time numbers only.** The stack is fully linked and started in code, but this build has not run on hardware yet, so runtime heap, CPU load and the mDNS interaction (CHIP uses the IDF `mdns` component the firmware's `ESPmDNS` also owns) are still unmeasured.

Three build-system items were needed to get it linking under pioarduino 3.3.11, all in `platformio.ini` plus one header, none in the component itself: `-DEXECUTABLE_COMPONENT_NAME=src` (esp_matter assumes the app component is `main`), a force-included shim with the same-type `operator==` for CHIP's ClosureControl structs (GCC 14 + gnu++2b; Espressif's newer Arduino cores inject `matter_closure_patch.h` for the same reason), and `-fexceptions` for `esp_modem` in the managed tree.

🤖 Generated with [Claude Code](https://claude.com/claude-code)