OpenEVSE/openevse_esp32_firmware #782: MQTT Causing OpenEVSE unit to reboot constantly

jeremyakers, 2024-01-02:
Hello,

I purchased an OpenEVSE for the specific purpose of using the MQTT functionality for solar divert EV charging.

However I am facing an issue that anytime MQTT is enabled: The EVSE constantly reboots itself every few seconds to a few minutes. This causes charging to stop and resets the "Smoothed current" back to 0. After a few seconds "Smoothed current" will climb back above 6 amps and charging will resume only to have the unit reset again which stops charging again, etc.

This functionality worked fine the first few weeks we owned the EVSE. We haven't changed any configurations. I did update to the latest Wifi firmware (v5.0.2) a day or so ago in an effort to correct the problem, but that was the only significant change I made and it was done well after the issue began.

I have played around with the settings and the unit resetting itself seems to occur anytime MQTT is connected. Regardless of whether "Self production" is enabled or not. If I disable MQTT the unit becomes stable. If I enable MQTT the unit starts constantly rebooting.

You can see in this screenshot an example of the resets:

[image]

It seems to add a new line to the history tab each time it resets even though the status is the same as the previous status. If I disable MQTT these duplicate status lines stop being added to the history tab. I can also see the LCD on the front of the EVSE turning off, coming back on and flashing the Wifi version and IP address. These resets also result in constant "Websocket disconnected" errors showing up in the UI.

I posted about the issue in the OpenEVSE support forums here: https://openev.freshdesk.com/support/discussions/topics/6000069051

But I was instructed to bring the issue here.

jeremyakers, 2024-01-03:
The MQTT broker is the one built into Solar Assistant (Which is based on Linux so I'm guessing Mosquitto) 

No secure connection. The MQTT broker is on my local network.

The issue does happen if Self Production is disabled as I described here:

> I have played around with the settings and the unit resetting itself seems to occur anytime MQTT is connected. Regardless of whether "Self production" is enabled or not. If I disable MQTT the unit becomes stable. If I enable MQTT the unit starts constantly rebooting.

Anytime the EVSE is connected to MQTT it starts resetting constantly. Here's a screenshot of what my MQTT settings look like:

[image]

And here's what it looks like if I use the mosquitto_sub command line client to subscribe to the broker and topic where the messages are published: (It's a cloudy morning right now so solar values are quite low):

```
$ mosquitto_sub -h 192.168.0.88 -u solar -P <password> -v -t 'gp2/goplug/1/chrgpwr/state'
gp2/goplug/1/chrgpwr/state 591.0
gp2/goplug/1/chrgpwr/state 591.0
gp2/goplug/1/chrgpwr/state 591.0
gp2/goplug/1/chrgpwr/state 591.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 588.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 585.0
gp2/goplug/1/chrgpwr/state 582.0
gp2/goplug/1/chrgpwr/state 582.0
```

The reason for the topic being named "goplug" is I was using this setup with a GoPlug EVSE before switching to OpenEVSE. Never had any issues with this setup on the GoPlug. When i switched over I just kept the same topic name.

jeremyakers, 2024-01-03:
Looking at $SYS/broker/version looks like the broker version is mosquitto 1.5.7:

```
$ mosquitto_sub -h 192.168.0.88 -u solar -P <password> -v -t '$SYS/broker/version'
$SYS/broker/version mosquitto version 1.5.7
```

chris1howell, 2024-01-05:
@jeremypoulter Any idea what may be causing this behavior? It is strange as MQTT / mosquitto / solar divert is a very popular combination. I have not seen any similar reports.

jeremypoulter, 2024-01-11:
Sorry, I haven't had a chance to look at this yet.

KipK, 2024-01-11:
Hi,

Seeing your configuration screen, I see you have set your base topic to
gp2/goplug/1

But your logs show you're using the same base topic for publishing your
solar data.
Evse base topic needs to be used only by OpenEVSE to publish its own data
or getting commands.

I'm not sure as I haven't crawled the codebase for a while, but can you
test to set another base topic, like "openevse/" ( as you're already
publishing the LWT in openevse/announce" why not using it for openevse data
too? )

jeremyakers, 2024-01-18:
Thanks @KipK for the response.

As an update: Over the weekend we went out of town and when we returned the EVSE wouldn't charge the car at all. When I tried to connect to it's local IP, I couldn't connect. Then I noticed that it was broadcasting it's own WiFi SSID. So I connected my smartphone to that SSID and found it seemed to have been factory reset somehow. It wanted me to go through the first time setup again. And apparently the default out of the box configuration is to default to "Disabled" which makes absolutely literally zero sense. Once I reconfigured that I could at least get the thing to charge my car. I have to say for a $600 piece of equipment I'm a little frustrated by the number of issues I've faced. I'm certainly an above average user when it comes to troubleshooting. A less technical EV driver would likely be absolutely confounded by these sorts of issues. Issues with MQTT are one thing... But having it default to "disabled" and having it randomly factory reset itself is another level of un-user friendliness.

I just reconfigured MQTT and so far I haven't noticed any resets... I tried it both the old way with the base topic set to "gp2/goplug/1/" as well as changing it to "openevse/" and no resets with either one.

Seems like the factory reset did something... But now my concern is: What caused it to factory reset on it's own and will THAT problem continue to re-occur? Probably absolutely no way of knowing but just wanted to share my experience.

I'll keep testing the solar divert now after this reset and report back if it starts resetting itself again.

chris1howell, 2024-01-18:
I agree an a out of the box default of "Disabled: is absolutely stupid and causes unnecessary headaches. I submitted an issue here:

https://github.com/OpenEVSE/openevse_esp32_firmware/issues/753

dnelson, 2024-03-15:
I'm having trouble with constant reboots too. I'm using MQTT and I think it started happening when I started trying to use claims to load-balance between my dryer and my EVSE, as both are on the same circuit. I'm trying to do more rigorous testing to narrow down the cause, but I'm having trouble creating a reproducible test setup. 

I'll follow up if I figure out anything, but I'd love to hear how best to debug. I'm open to building a test rig or hooking up a debug connection, I just need to figure out what direction to go down.

rnavarro, 2024-04-09:
I'm experience these same issues on:

OpenEVSE 8.2.2
OpenEVSE Wifi v5.1.0_modified (It's the .bin from this page: https://github.com/OpenEVSE/openevse_esp32_firmware/releases/tag/v5.1.0)

closed 2026-09-21 by github-actions[bot]