OpenEVSE/openevse_esp32_firmware #783: Wifi Bug ? Open EVSE 4

Nick425321, 2024-01-06:
I have two OpenEvse's one is a v4.0 and the other is a v5.0
The v5.0 has 8.2.3 firmware and 5.0.2 wifi firmare.
The v4.0 has 8.2.3 firmware.

I had both units on the wifi dev firmware and down graded to 5.02.
The v5.0 works normally.
The v4.0 stays at "Waiting to Initialize" so I installed a spare ESP32 that I have with firmware 4.2.2 and it booted normally to the access point.
Once I updated the WiFi settings to connect to my access point and rebooted after powering down to replace the cover. I now get "Waiting to Initialize"

I will down grade the v4.0 to 8.2.2 so it is usable.

chris1howell, 2024-01-06:
8.2.3 is the problem. It was compiled by a developer with Boot lock enabled
which locks the station until WiFi boots up fully and unlocks.

This is a big problem if your station does not have WiFi.  To compound the
issue even if you do have WiFi only one older WiFi version also has boot
lock enabled.

I have submitted an issue with to add RAPI commands to enable disable boot
lock. And added warnings on 8.2.3.

Unless you need bootlock and have a compatible version you should not
install 8.2.3.

Nick425321, 2024-01-06:
I have dropped down to 8.2.2 firmware and it now works using wifi 5.0.2.
The strange thing is on the v5.0 unit with 8.2.3/5.0.2 all 3 ESP32's boot fine.

closed 2026-09-21 by github-actions[bot]