OpenEVSE/openevse_esp32_firmware #788: TransactionMessageAttempts and TransactionMessageRetryInterval is get status "Not Supported"

flyrobot, 2024-01-15:
"key":"TransactionMessageAttempts","value":"0" 
"status":"NotSupported"

"key":"TransactionMessageAttempts","value":"3"
"status":"NotSupported"

"key":"TransactionMessageRetryInterval","value":"60"
"status":"NotSupported"

is any specific value needed ?

matth-x, 2024-01-15:
The OCPP library does not support these configs, that's true. Currently, the MessageAttempts are unlimited and the RetryInterval is a constant. Does this affect the compatibility with your backend?

flyrobot, 2024-01-16:
Yes this is a mandatory to comply in backend. they also have some charger and i see they do not have this issue.

matth-x, 2024-01-16:
Add custom configs like this:

https://github.com/OpenEVSE/openevse_esp32_firmware/blob/f4a3d3114b5d1e37c1d66568bd710d024b4a90e3/src/ocpp.cpp#L107-L112

After these lines, add new `declareConfiguration` instructions. Change `<bool>` to int and pass any factory default values.

The OCPP lib will ignore these configs, but I assume that the tests won't verify the behavior.

flyrobot, 2024-01-17:
ArduinoOcpp::configuration_init(filesystem); 
 ArduinoOcpp::declareConfiguration<const char*>( 
     "MeterValuesSampledData", "Power.Active.Import,Energy.Active.Import.Register,Current.Import,Current.Offered,Voltage,Temperature"); 
 ArduinoOcpp::declareConfiguration<bool>( 
     "AO_PreBootTransactions", true); 
ArduinoOcpp::declareConfiguration<int>( 
     "AO_PreBootTransactions", true); 

do you mean like become like this ?

flyrobot, 2024-01-18:
i try this config above, still doesn't work. may be im wrong add a new declareConfiguration, please advice.

closed 2026-09-21 by github-actions[bot]