OpenEVSE/openevse_esp32_firmware #790: Get Diagnostics, always upload failed

flyrobot, 2024-01-16:
Sending: [2,"48496111-3ac1-4461-8b17-9c224a5e7b24","GetDiagnostics",{"location":"C:"}]
Received: [3,"48496111-3ac1-4461-8b17-9c224a5e7b24",{"fileName":"diagnostics.log"}]
Received: [2,"1000003","DiagnosticsStatusNotification",{"status":"UploadFailed"}]

matth-x, 2024-01-16:
Get Diagnostics works for http:// URLs only at the moment. If you send the location parameter "http://example.com/my/diagnostics.log", then OpenEVSE will initiate an HTTP upload to that server to the path "/my/diagnostics.log".

flyrobot, 2024-01-16:
ok thanks, i will try

flyrobot, 2024-01-17:
"GetDiagnostics",{"location":"http://localhost:1180/"}
{"fileName":"diagnostics.log"}
"DiagnosticsStatusNotification",{"status":"Uploading"}

this is what we got status uploading and same as reported from test result

matth-x, 2024-01-21:
Try to figure out the correct upload URL with a second computer first. I.e. your test HTTP server is on computer 1, and the test client is computer 2 (both are in the same LAN). 

If you're able to upload a file from computer 2 to 1, then the same URL will work for OpenEVSE too.

closed 2026-09-21 by github-actions[bot]