OpenEVSE/openevse_esp32_firmware #791: GUI not load

vohum, 2024-01-16:
`Failed to load module script: Expected a JavaScript module script but the server responded with a MIME type of "text/html". Strict MIME type checking is enforced for module scripts per HTML spec.`

probably after updating vite device can't load .js files

maybe it's simmilar new issue
https://stackoverflow.com/questions/75433591/failed-to-load-module-script-expected-a-javascript-module-script-but-the-server

jeremypoulter, 2024-01-16:
Is this with a local build?

vohum, 2024-01-17:
https://github.com/OpenEVSE/openevse_esp32_firmware/blob/f4a3d3114b5d1e37c1d66568bd710d024b4a90e3/src/web_static/web_server_static_files.h#L22

I know the reason in those lines code changed to `/assets\components-3194f1c3.js` I don't know why

closed 2024-01-17 by vohum

KipK, 2024-01-17:
Each time you modify something on UI file, Vite will generate a new hash for this file to prevent caching issues.