OpenEVSE/openevse_esp32_firmware #793: Feature Request: IPv6 support

njh, 2024-01-18:
Hello,

I prefer to use IPv6 where possible. My [ESP32-Faikin](https://github.com/revk/ESP32-Faikin) module supports it.
Is it something that could be added to OpenEVSE?

Thanks,

nick.

apalrd, 2024-03-25:
+1 to firmware support for v6 addressing. ESP-IDF has good v6 support already.

njh, 2024-03-25:
Hm, this might be the issue; ESP-IDF has IPv6 support but OpenEVSE is built on Arduino and I am not sure how good the IPv6 support is on that.

Although having checked, there is an `enableIPv6()` method!
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.h#L97

Looks like a whole lot of work was merged in January:
https://github.com/espressif/arduino-esp32/pull/9016