OpenEVSE/openevse_esp32_firmware #798: “Waiting for initialization” after power loss

TechnicalLee, 2024-01-23:
If power is interrupted or the OpenEVSE board is rebooted, the board will not exit boot lock and displays “Waiting for initialization” until the charge status is manually toggled to disable charge first and then enable charge (the default state after reboot is AUTO charge). The GUI works fine the whole time. I have the 5.0.2 firmware, didn’t have this issue in 4.x. Main board is 8.2.3 firmware.

It would appear the WiFi firmware is not sending the correct wake up signal automatically after a power cycle or OpenEVSE board reboot.

chris1howell, 2024-01-30:
8.2.3 is the problem. It was compiled by a developer with Boot lock enabled
which locks the station until WiFi boots up fully and unlocks.

This is a big problem if your station does not have WiFi.  To compound the
issue even if you do have WiFi only one older WiFi version also has boot
lock enabled.

I have submitted an issue with to add RAPI commands to enable disable boot
lock. And added warnings on 8.2.3.

Unless you need bootlock and have a compatible version you should not
install 8.2.3.

dmelliott20, 2024-03-26:
Having the same issue here. Only noticed the issue when upgrading to 5 too

SergioRius, 2024-04-06:
I'm having to open the box to physically push the reset button on the openevse after a blackout, and I'm on 8.2.2EU.
The screen shows a line of solid blocks after the power loss. I don't know if it's the same issue.

PR #888 (2024-08-08): Be more agressive about clearing the boot lock
There was a check to see if the EVSE was not in the starting state, but it does not exit starting until the boot lock in cleared. Just removed that check as it is not needed as there is a flag to indicate if the boot lock is set.

Fixes #798 “Waiting for initialization” after power loss
commit b2662971 (2024-08-08): Be more agressive about clearing the boot lock

There was a check to see if the EVSE was not in the starting state, but it does not exit starting until the boot lock in cleared. Just removed that check as it is not needed as there is a flag to indicate if the boot lock is set.

Fixes #798 “Waiting for initialization” after power loss
closed 2024-08-08 by jeremypoulter