OpenEVSE/openevse_esp32_firmware #805: ESP32-Gateway No RJ45 Ethernet Using V5.1.0

rbhollabaugh, 2024-02-27:
Upgraded from 5.0.1 to 5.1.0 using the web page upgrade firmware option. Hard wired ethernet stopped working.  No wifi in the garage.

Esp32 gateway board is revision G

OpenEvse Ardiuno firmware 8.2.2

Using USB cable installed 5.0.2 and is now working again.

stben67, 2024-02-28:
Good morning

I have the same problem after updating to version v5.1.0.
the ethernet connection no longer works.

Can you help us

SergioRius, 2024-03-18:
The same problem here. I just came from the garage to the desktop with the boards.

jeremypoulter, 2024-03-20:
There was no firmware binary released for the ESP32-Gateway, so not sure why it is being offered as an update... I will add these binaries to the release

rbhollabaugh, 2024-03-21:
Hi Jeremy - Flashed the 5.1 esp32 gateway file you just added (14 hours ago). It's the same filesize as the one that was there that did not work. And this one also has the same problem. Going back to 5.0.2

Tinbum1, 2024-03-23:
Ditto, exactly the same problem.

stben67, 2024-04-07:
The problem is not solved. Can you check the 5.1 esp32 gateway file ?

Thanks

closed 2024-05-08 by jeremypoulter

SergioRius, 2024-07-03:
> It is likely to be [this](https://github.com/OpenEVSE/openevse_esp32_firmware/commit/dbb3eddeb6a58f4cccaa698c0655d15e5ff85422) update to the Arduino core that caused the issue, the ethernet port is no longer started up correctly.

Any idea in when this commit will be deployed?

I'm having an issue and I would want to know if it also reproduces on a newer version.
If you have a Soc limit set, nightly scheduled charge executes with the car at a higher Soc, so it doesn't charge, it seems to get "disabled". Then during the day you change the limit to a higher one so it charges from sun, but the charger is non responsive. The toggle buttons are disabled and the charger doesn't work. The only way of making it work again is restarting the gateway.

Thank you.

PR #827 (2024-05-08): Library updates
Fixes #805 and build errors