OpenEVSE/openevse_esp32_firmware #809: Wrong consumption values on 3-phase connection

vermut, 2024-03-07:
OpenEVSE reports consistent set of metrics, but they don't line up with other sensors outside OpenEVSE. It's like it's using max current for calculations instead of actual current.

### OpenEVSE world
[image]

[image]

2.5h * 21A * 220V * 3 phases = ~34kWh

21A is the limit set in OpenEVSE. Everything is consistent with each other.

### Everything else
I'm charging RAV4 plugin that has battery size 18 kWh. Power sensors on solar inverter and data from electric company roughly match each other and say that during that period only  ~7kWh was consumed. They may be inaccurate but not THAT inaccurate.

[image]

And my **daily** consumption (those are accurate from electric company) is never even reaching this value.
[image]

---

OpenEVSE: 	8.2.3.T2 	
OpenEVSE Wifi: 	v5.1.0_modified
[image]

zerodur12, 2024-04-06:
the RAV4 on board charger (AC) is a 6.6kW - 28A/230V **1 phase** Maximum charging power AC.

Even you have a 3 phase charger you will charge using only on 1 phase. 
in 3 phase OPENEVSE Charger the current is mesure only on 1 phase, it compute the power on 1 phase and multiply the result by 3 assuming the 2 other phase are loaded with the same current. 

1 your case 2 phases are unloaded the repported power is wrong and has to be divide by 3...

vermut, 2024-04-21:
If it's one phase, maybe the solution would be switch OpenEVSE to 1-phase mode?

vermut, 2024-04-30:
I did that.

closed 2024-04-30 by vermut