OpenEVSE/openevse_esp32_firmware #812: Timers over rides Eco mode

400bird, 2024-03-16:
With anything set in a timer, Eco mode doesn't work.

I'd like to have both working together.

An active timer was the root cause of this issue:
https://github.com/OpenEVSE/openevse_esp32_firmware/issues/691

rnavarro, 2024-03-29:
+1 to this, i'm experiencing the same issue

400bird, 2024-03-29:
I also opened a ticket on the OpenEVSE support forum. I think the problem is now better understood and a long term solution was proposed.

Until that is implemented (if ever), deleting all the timers allowed eco mode to function normally. It's not ideal because now I have to rely on timers in the car, which are annoying to disable when I really need to charge now and skip the timers.

https://openev.freshdesk.com/support/tickets/85079

gbtucson, 2025-09-30:
I would also like to see timers working with ECO mode - very simply, a timer lets me block charging during peak periods which I want with ECO (as well as all other) modes because it's easier just to program and forget about it.
ECO mode reacts slowly enough so that when my house a/c kicks in, I'm always dipping into peak time utility rates for a few minutes before ECO drops down enough to balance. In the summer this can happen every few minutes, meaning the ECO mode is always runing behind and I pay for it...
Much simpler just to have a timer stop it.... then I can really forget about it.

closed 2026-09-21 by github-actions[bot]