OpenEVSE/openevse_esp32_firmware #818: UI issue with client ID 4 claim

fhteagle, 2024-04-01:
When Home Assistant OpenEVSE integration (by @firstof9 ) makes a claim, that claim ID is tagged to client ID 4. Then, the web UI shows the automation logic to be driven by "Clients.Null" . Was this client ID not given a correct friendly name to display?

[image]

[image]

firstof9, 2024-04-01:
This shows "null" as I haven't requested a proper ID to be added to the OpenEVSE firmware at this time.

fhteagle, 2024-04-01:
Possibly related to https://github.com/OpenEVSE/openevse_esp32_firmware/issues/536  then?

firstof9, 2024-04-01:
Basically, yes

closed 2026-09-21 by github-actions[bot]