OpenEVSE/openevse_esp32_firmware #822: Confusing OpenEVSE Wifi version

dml33, 2024-04-05:
Hi.

I have just updated with the huzzah32 binary from 5.0.2 and I have just noticed that the OpenEVSE Wifi version is quite confusing. Instead of 5.1.0, the github hash `master_d92677f6_modified` is shown:

[image]

So, even having the 5.1.0, it shows as if it not already updated...

[image]

In case of the wifi_v1, the name shown is ` v5.1.0_modified`

I think it would be nice to have a clear name, like it was in 5.0.2: `v5.0.2`

jeremypoulter, 2024-04-05:
Oh right, I know why. The release GUI module was not updated properly before the release, and then the missing binary was uploaded from the wrong source. I will do a new release

SergioRius, 2024-04-06:
> Oh right, I know why. The release GUI module was not updated properly before the release, and then the missing binary was uploaded from the wrong source. I will do a new release

I hope that also fixes the lack of ethernet for the Olimex. I'm waiting on 5.0.1 (8.2.2EU) but it has important issues, like having to open the box to manually reset the openevse after a blackout or the shaper not working. :"(

jeremypoulter, 2025-08-27:
this has been resolved in later releases