OpenEVSE/openevse_esp32_firmware #823: Solar Divert should also work when Current Measurement does not

pdhoogh, 2024-04-10:
This is one for Guillaume :-).

The current measurement is out on my OpenEVSE and the divert no longer properly works. The pilot is jumping up and down all the time. That may be because the divert algo uses the measured current. If I'm right, than I should say the EVSE measured current really shouldn't be in the algo. The control function here is to move the pilot (the manipulated variable) to get +I/-E (the process variable) to a setpoint of zero. The measured charge current does not matter.

For all I know, using the measured charge current introduces error. Because the car charger is controlling the charge current, not the EVSE. The car charger responds to the pilot if it wants to. And if the EVSE measured current does not match the car charger's measurement or allowance, that can only cause problems.

Of course you are right to say "fix the CT", but hey, isn't graceful degradation a beautiful thing?

If you want I can help with explaining the PID control algo. The trick is to make it incremental and not full value. Assume the output (pilot) is X. Then look at the error of the process variable vs. setpoint, i.e. how much is +I/-E away from zero. That is DeltaY. Calculate DeltaX as a function of DeltaY. The new output is X + DeltaX. You could use an integral only function, DeltaY = factor * DeltaX. "factor" is is always <1 and is a function of the integral time here, but remember that the factor here is also a function of sample period if we want to use true time.

lutorm, 2024-07-01:
I don't know how the current code works, but it seems the problem with just steering pilot based only on available power is that the vehicle may not pull all that it is allowed. Ie, a situation where there's 5A available and the vehicle is drawing 4A is fundamentally different from a situation where there's 20A available and the vehicle is drawing 19A. In the first case, the control should never set the pilot higher than 5A, in the second never higher than 20A, but using a PI controller without knowledge of the actual current would result in the pilot winding up to whatever arbitrarily large current the controller is limited to. This would be very bad if another vehicle is connected that suddenly can draw all that current, so would exceed the allowable power and potentially tripping the main breaker.

pdhoogh, 2024-07-01:
Indeed the on-board car charger is master of its faith. Under normal circumstances it will never pull less than 6A as this is the minimum current allowed for EV charging.

I also was not sufficiently clear about the PI controller. Obviously, if the true current "process value" is zero because the measurement is broken, it will always wind up to max. My reason for including the controller algo was to provide an alternative to the current code that uses the measured current and the available current to move the pilot. Of course that no longer works if the measured current is out.

Anyway, this is not so important. I purchased a new control board from OpenEVSE after some tests pointed to a failed chip. Everything works fine again.

closed 2026-09-21 by github-actions[bot]