OpenEVSE/openevse_esp32_firmware #824: Self-production value reading issue

doppiaemme, 2024-04-12:
v5.1.0

When Self-production is set to "Mode Production" the feed value is read correctly
When Self-production is set to "Export" the feed value is stuck to a certain value and never changes.

Feed MQTT topic is still the same

Am I doing something wrong?

In the past I was using Site Export with a negative value to modulate the charging but now in 5.1.0 doesn't seem to work anymore for me.

thekev, 2024-06-28:
same here
[image]
[image]

5var0g, 2025-01-17:
I had the same problem. The solution is to have different "Feed" for "Production" and "Export".

closed 2026-09-21 by github-actions[bot]