OpenEVSE/openevse_esp32_firmware #825: [RFC] adding a voltmeter

kolyshkin, 2024-04-16:
_This is a continuation/summary of discussion in https://github.com/lincomatic/open_evse/issues/175_

Current OpenEVSE hardware (as of April 2024) lacks a voltmeter, and the current meter is not too precise. As a result, power calculations use some predefined voltage (like 240) and imprecise current.

After some research, it seems that adding a dedicated power meter that will interface with ESP32 directly (i.e. no ATMEGA involvement) would be the best solution.

For example, Peacefair PZEM-004t (v3.0) measures both AC voltage and current (up to 100A) and is able to talk via a serial bus. There are mixed reports of whether it can work with ESP32's 3.3v signalling, but most sources say it can (if being powered by 3.3v).

Another possibility is to use Peacefair PZEM-016 which seems to be a newer version of PZEM-004t v3.0 (as far as I can tell, the only difference is built-in power supply that is powered from the high voltage side).

ESP32 has 3 serial ports. Current OpenEVSE hardware (OpenEVSE TFT v1) has two serial ports available on the board:
 - Serial0 (used to talk to ATMEGA board);
 - Serial2 (seems to be used for development/debug, although the connector is not soldered).
 
 There's also a possibility to use Serial 1 by directly connecting to any two ESP32 pins that are otherwise unused.
 
For now, I am going to try using Serial2 (aka debug port). Unfortunately it is used even when the `DEBUG` is not defined, so the first step would be to make debug work without actually occupying a real serial port.

The next step would be to modify the code to support a dedicated power meter instead of relying on information from ATMEGA board.

Any ideas, comments, questions etc. are very welcome!

Dicion, 2024-06-19:
Looks like the PZEM-004t is RS232 on the serial side, and PZEM-016 is RS485.

It would be REALLY easy if someone made an i2c version of something like this. Can just bus it on the QWIC connector along with the RFID reader.

It might be worth looking into using something like this: https://www.dfrobot.com/product-2001.html

Especially since you're having difficulty accessing the serial ports already.

Dicion, 2024-06-19:
Assuming you're using the TFT board, you could modify platformio.ini to move the debug port to the unsoldered one

From

[env:openevse_wifi_tft_v1]
board = denky32
build_flags =
  ${common.build_flags_openevse_tft}
  -D DEBUG_PORT=Serial2
  -D RAPI_PORT=Serial

To

[env:openevse_wifi_tft_v1]
board = denky32
build_flags =
  ${common.build_flags_openevse_tft}
  -D DEBUG_PORT=Serial1
  -D RAPI_PORT=Serial

Which would free up Serial2 for your use, and if you ever needed the debug, you could move it back, or solder it to the ESP32 pins.

closed 2026-09-21 by github-actions[bot]