OpenEVSE/openevse_esp32_firmware #829: MQTT No Longer Connects after upgrade to v5.1.0_modified

jeremyakers, 2024-04-29:
OpenEVSE is refusing to connect to my MQTT broker even after powering down and restarting the unit:

[image]

And yet I can connect from my local PC on the same network just fine:

```
$ mosquitto_sub -h 192.168.0.88 -u solar -P <password> -v -t 'gp2/goplug/1/#'
gp2/goplug/1//override {"state":"null"}
gp2/goplug/1//claim {"state":"null"}
gp2/goplug/1//schedule []
gp2/goplug/1//limit {"type":"none","value":0,"auto_release":true}
gp2/goplug/1//config {"mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http"],"buildenv":"openevse_wifi_v1","version":"v5.0.2","wifi_serial":"B0A732FDCF04","protocol":"-","espinfo":"ESP32r3 2 core WiFi BLE BT","espflash":16777216,"firmware":"8.2.2","evse_serial":"M06SBEFF0B787B","diode_check":true,"gfci_check":true,"ground_check":true,"relay_check":true,"vent_check":true,"temp_check":true,"max_current_soft":40,"service":2,"scale":220,"offset":0,"min_current_hard":6,"max_current_hard":80,"ssid":"dlink-4D70","pass":"_DUMMY_PASSWORD","ap_ssid":"","ap_pass":"","lang":"","www_username":"","www_password":"","hostname":"openevse-cf04","sntp_hostname":"pool.ntp.org","time_zone":"America/Chicago|CST6CDT,M3.2.0,M11.1.0","limit_default_type":"","limit_default_value":0,"emoncms_server":"https://data.openevse.com/emoncms","emoncms_node":"","emoncms_apikey":"","emoncms_fingerprint":"","mqtt_server":" 192.168.0.88","mqtt_port":1883,"mqtt_topic":"gp2/goplug/1/","mqtt_user":"solar","mqtt_pass":"_DUMMY_PASSWORD","mqtt_solar":"gp2/goplug/1/chrgpwr/state","mqtt_grid_ie":"emon/emonpi/power1","mqtt_vrms":"","mqtt_live_pwr":"","mqtt_vehicle_soc":"","mqtt_vehicle_range":"","mqtt_vehicle_eta":"","mqtt_announce_topic":"openevse/announce/cf04","ocpp_server":"","ocpp_chargeBoxId":"","ocpp_authkey":"","ocpp_idtag":"DefaultIdTag","ohm":"","divert_type":0,"divert_PV_ratio":1.1,"divert_attack_smoothing_time":20,"divert_decay_smoothing_time":20,"divert_min_charge_time":600,"current_shaper_max_pwr":0,"current_shaper_smoothing_time":60,"current_shaper_min_pause_time":300,"current_shaper_data_maxinterval":120,"vehicle_data_src":0,"tesla_access_token":"","tesla_refresh_token":"","tesla_created_at":18446744073709551615,"tesla_expires_in":18446744073709551615,"tesla_vehicle_id":"","rfid_storage":"","led_brightness":128,"scheduler_start_window":600,"flags":115344906,"emoncms_enabled":false,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"mqtt_retained":false,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"current_shaper_enabled":false,"pause_uses_disabled":false,"mqtt_vehicle_range_miles":false,"ocpp_enabled":false,"ocpp_auth_auto":true,"ocpp_auth_offline":true,"ocpp_suspend_evse":false,"ocpp_energize_plug":false,"rfid_enabled":false,"factory_write_lock":true,"is_threephase":false,"wizard_passed":true,"default_state":true,"mqtt_protocol":"mqtt","charge_mode":"eco"}
gp2/goplug/1//config_version 1
gp2/goplug/1/chrgpwr/state 7957.5
gp2/goplug/1/chrgpwr/state 7957.5
gp2/goplug/1/chrgpwr/state 7957.5
gp2/goplug/1/chrgpwr/state 7957.5
gp2/goplug/1/chrgpwr/state 7957.5
gp2/goplug/1/chrgpwr/state 7955.0
gp2/goplug/1/chrgpwr/state 7955.0
gp2/goplug/1/chrgpwr/state 7955.0
gp2/goplug/1/chrgpwr/state 7955.0
gp2/goplug/1/chrgpwr/state 7955.0
gp2/goplug/1/chrgpwr/state 7955.0
```

I have tried disabling and re-enabling MQTT and self production, restarting the unit, etc. Nothing seems to work.

zerodur12, 2024-05-06:
may be the MQTT problem is relate to  the bug see "Let's Encrypt CA is out of date" #806 if yes the workaround is to accept self-signed certificates

jeremyakers, 2024-05-13:
I do not have SSL enabled on my MQTT broker as it runs on my local network.

This issue eventually just went away on it's own. Several days later the EVSE finally successfully connected to the MQTT broker.

njh, 2024-08-26:
I think I also just experienced this. Pressing the 'Restart' button OpenEVSE Wifi in the Firmware settings fixed it.
I have an ESP32 Ethernet connection and running Wifi Firmware version `v5.0.2`.

pbix, 2024-11-16:
I am running v5.1.2 and this issue still persists.  My observation is that is openevse get disconnected from the broker for example there is transient loss of Wifi signal then it will not reconnect.  The only way I have found to make it reconnect is to restart the Openevse Wifi module.  I ran for many years on v4 and never experienced this.

closed 2026-09-21 by github-actions[bot]