OpenEVSE/openevse_esp32_firmware #849: Network library update

jeremypoulter, 2024-05-21:
We are still using Mongoose 6 and while this mostly works I have seen many issues while trying to implement #29 and #806. A lot has changed since we originally chose Mongoose so we should evaluate different options.
- Mongoose 7 #35
- ESPAsyncTCP #354
- native ESP IDF
- [PsychicHttp](https://github.com/hoeken/PsychicHttp)

jeremypoulter, 2024-05-21:
@matth-x do you have any input on what might be a good option?

jeremypoulter, 2024-05-21:
LWS looks interesting, also includes a mini HTML/CSS display engine targeted at the ESP32, could be great for the new LCD WiFi board, will definitely take a look.

razvanphp, 2025-05-20:
what about [Zephyr Project](https://www.zephyrproject.org/zephyr-rtos-on-esp32/)?

closed 2026-09-21 by github-actions[bot]