OpenEVSE/openevse_esp32_firmware #856: OCPP - GetCompositeSchedule returns unknown variable?

retrocoder-78, 2024-06-07:
Hello all!

Been testing the OCPP side and behaviour compared to other OCPP units. Today was charging at 16amps set in the openevse firmware. Was unsure as to why (pretty sure it was to be set to 32 amps) so though I would do a steVe GetCompositeSchedule to see how the station is currently set.

It returned the screen. Usually if the unit is set to let's say, 24 amps, the OCPP command will return 24amp if requesting the information in amps.

@matth-x Any thoughts?

Thank you for your time! :)

[image]

matth-x, 2024-06-10:
Hi @retrocoder-78,

Thanks for sharing this finding. The OCPP client actually only reports the load limits defined by OCPP Charging Profiles and the value -1 means that no limit is defined. MicroOcpp has no interface to also report the schedule from the local load balancer or other load management services (just hasn't been requested yet).

To monitor the maximum charge rate as advertised by the control pilot, you can add the measurand "Current.Offered" to the list of MeterValuesSampledData.

Do you have further findings where the OpenEVSE behaves different from other chargers?

retrocoder-78, 2024-06-13:
Great information! The value of -1 and the explanation makes sense. I am testing some other popular OEM hardware at the moment. I'll report back shortly as to the return of the GetCompositeSchedule request and it's result.

Cheers! :smile:

retrocoder-78, 2024-06-19:
Hello!

My apologies for the delay. The OEM charge box I am testing is fairly common in the market place. It's manufactured by LITEON Technology. They are rebranded by many companies.
[image]

When no OCPP charge profile is loaded, the GetCompositeSchedule command returns the current Control Pilot setting. This box ships by default set at 40 amps. So, this is what the command returned.

[image]

Taking a look in the firmware, there's an area to check what OCPP charge profiles might be loaded.

[image]

This is a nice feature, as once charge profiles are loaded via SteVe, keeping track of this and their stack can be problematic.

[image]

Okay, so that's the report back from inside this very common charge box. 

If there's anything further I can report back on, please don't hesitate to let me know. 😄

Cheers! 

> Hi @retrocoder-78,
>
> Thanks for sharing this finding. The OCPP client actually only reports the load limits defined by OCPP Charging Profiles and the value -1 means that no limit is defined. [...]

matth-x, 2024-06-20:
Thank you for the further details! I will consider reporting the default limit in the Composite Schedule, because it really makes sense to me.

retrocoder-78, 2024-06-20:
Amazing! Glad this helped the cause. 

Cheers!

> Thank you for the further details! I will consider reporting the default limit in the Composite Schedule, because it really makes sense to me.

closed 2024-06-20 by retrocoder-78