OpenEVSE/openevse_esp32_firmware #866: Feature request: Password-protected current limit

swansontec, 2024-06-21:
According to the NEC, an EVSE installer needs to size the breaker & wires according to the nameplate rating, or they can use a smaller breaker & wires if the lower current setting is either:

- Only settable by physically opening the unit with a tool or key
- Only settable by password-protected commissioning software

See 625.42 of NEC 2023.

Since the OpenEVSE firmware has no password protection on the current-limit setting, forces us to use the nameplate rating of 48A. Anything else would be a code violation. This makes it illegal to use the NEMA 14-50 plug included in the kit, since that requires a 40A current limit (40A x 125% = 50A).

The simplest way to solve this would be to add a "commissioning password" that only applies to the settings on the "EVSE" page. These settings, such as the current limit, 3-phase vs 2-phase, level 1 vs. level 2 charging, etc., all depend on the installed hardware, and generally shouldn't be changed once the unit is physically assembled. It makes sense to configure these are part of the "commissioning" step, and then lock them from future changes. Other settings would remain user-configurable, such as the WiFi password, charging schedule, or API integrations.

Another option would be to install a physical jumper on one of the boards. Setting this jumper would lock the EVSE settings, preventing modification in software. To change the setting, one would have to physically unscrew the cover to remove the jumper. This option requires hardware changes, though, so that wouldn't work for existing owners.

Dicion, 2024-06-21:
So the actual, full, text of the NEC is as follows:

[image]

The EVSE comes, as sold from the store, out of the box, with the WiFi in AP mode, and requires a custom SSID and default password to access the user interface, detailed in the installation guide, and should only be installed by a qualified person, where you change and lower the setting.

So it is password protected out of the box, with the password only accessible to a 'qualified person' in the way of any other piece of hardware where the password is in the installation guide, and told to only be installed by 'qualified persons'.

Just in case you think this means 'only an electrician should be able to change this', that is not the case. The NEC Definition of 'Qualified person' is:

"The National Electrical Code (NEC) defines a qualified person as someone who has the skills and knowledge to construct and operate electrical equipment and installations. They must also have received safety training to identify and avoid hazards."

It meets NEC Code requirements out of the box by requiring a password to access the web interface to set the current.

Dicion, 2024-06-21:
It meets NEC out of the box by requiring a password to initially connect to the device in AP mode to make any and all settings.

However, a password protected current change on the page isn't a bad idea. Maybe make it default to the last 4 of the mac address of the wifi controller or something that's different per unit, but easy to find/get.

You might want to post it over on the actual web ui repo as well: 

https://github.com/OpenEVSE/openevse-gui-v2/

jeremypoulter, 2024-06-22:
The OpenEVSE module has a write once current setting for this sort of thing, however we don't currently use this in the UI.

Dicion, 2024-06-22:
> The OpenEVSE module has a write once current setting for this sort of thing, however we don't currently use this in the UI.

That requires a full reflash to change again once set IIRC, correct?. This request is more, repeatedly settable, but with a restricted password/access ability. But yes, that certainly meets the 'tool access' requirement as well, as it requires opening the case to reflash the module every time you want to change it, and no one but a 'qualified person' will know how to reflash.

swansontec, 2024-06-24:
Thanks! I did not know about the write-once current setting. Is there documentation somewhere for that? I'm guessing that it involves editing the code and then flashing the customized firmware.

This setting would solve the immediate problem with my electrical inspection, although I still think the UI should separate "stuff only an installer should access" from "stuff my wife can access".

closed 2026-09-21 by github-actions[bot]