OpenEVSE/openevse_esp32_firmware #873: ocpp server connect fail

liyong4github, 2024-07-01:
I am unable to connect to the OCPP server. Can anyone help me identify the problem? Here is the error message:

[MO] warning (MicroOcppMongooseClient.cpp:261): Insecure connection (WS)
[MO] info (RequestQueue.cpp:45): operation timeout: BootNotification
[MO] warning (MicroOcppMongooseClient.cpp:485): connection ws://xx.xx.xx.xx/steve/websocket/CentralSysremService/test01 -- HTTP error 404
[MO] info (MicroOcppMongooseClient.cpp:511): connection ws://xx.xx.xx.xx/steve/websocket/CentralSysremService/test01 -- closed
OpenEVSE not responding or not connected

matth-x, 2024-07-01:
HTTP error 404 means that the charger can reach the server, but the server doesn't recognize the URL. The backend URL you posted contains a typo:

ws://xx.xx.xx.xx/steve/websocket/CentralSys**r**emService

liyong4github, 2024-07-03:
@matth-x I'm so stupid. Thank you so much.

closed 2024-07-03 by liyong4github