OpenEVSE/openevse_esp32_firmware #880: OCPP transaction error, Temperature limit hit.

retrocoder-78, 2024-07-12:
With the recent hot weather, openevse box hits the temperature limit of 71C. It's been in direct sunlight so moving to the shade fixes the heat issue. 

Connected to Steve, when the temperature limit is hit, the transaction that was current hangs and has to be manually stopped. 

What's the current thoughts on this use case? 

It seems, that the unit has faulted, so, the connector status would report such until the fault is cleared. 

And, that faulting of the unit, the transaction should be ended. This would avoid the hanging transaction in Steve. 

A soft reboot from Steve should reset the fault and the hardware is ready to attempt another transaction. 

Any thoughts or suggestions @matth-x would be great! 🤓

matth-x, 2024-07-12:
Hi @retrocoder-78, thank you for the report! The standard behavior is that the transaction is suspended, but as soon as the box cooled down again, charging would resume. During these steps, the OCPP status goes from Charging to Faulted to Charging again. So my first thought is that the transaction didn't hang, but was just suspended. Does this sound plausible?

The failure to start new transactions confuses me a bit. How did you start the transactions and how did you stop them (e.g. via RemoteStopTransaction)?

retrocoder-78, 2024-07-13:
Thank you for the run down of the logic in this scenario @matth-x . I think you're correct. 

I attempted to remotely restart the unit from Steve during the faulted state. I think this caused the hung transaction in Steve. From this point, the vehicle was still plugged in and the unit tried to start another transaction. But, with the current transaction still running, the box couldn't because the tagid was already in use. The LCD reported 'card unkown'. This lead to the DeAuthorized stop reason in Steve. I believe this is all correct, looking at the transaction data and series of events that occurred.

Once I manually looked at the specific transaction and stopped (Stop button) the transaction in Steve, the vehicle was able to start another transaction again.

The logic here makes sense. Allow the unit to cool down and then continue the charging session. And, the transaction is still live in Steve so all the data is logged during the over temperature conditions. Which was helpful to look back and see what was occurring.

Given this scenario, I wonder what improvements could be made if any? If the unit is sent either a soft or hard reboot during operation, does a command get sent to the backend to close the transaction? Or, is that transaction kept live during that reboot?

Appreciate your insight! :clap:

retrocoder-78, 2024-07-22:
I did some testing. And, yes, when the unit is in transaction, and a reset command is sent, the current transaction is closed. Since the vehicle is still plugged in, the unit continues the charging session but begins a new transaction in Steve. Something to note, the 'stop reason' stated is left blank. If I get any of the odd behaviour in the future, I'll reopen this issue. :smile:

closed 2024-07-22 by retrocoder-78