OpenEVSE/openevse_esp32_firmware #884: Openevse OCPP End point address updating via Steve when firmware access unavailable?

retrocoder-78, 2024-07-14:
Just looking for discussion and thoughts...

One of the major OEM evse manufactures (I know of) is working with network backend operators in this fashion, updating remotely the OCPP url via the backend which takes effect on the evse. If owners of many units per site decide to migrate backend services to a different backend operator, many units would need to be manually configured with the new OCPP url. This is potentially time consuming and by extension expensive.

Once first connection is done, firmware settings are set and a connection with Steve is established. If the WS (WSS) address ever needed to be adjusted without firmware access, it seems to me to be that,

1. A custom build of Steve would be required. From reviewing all the OCPP documentation, I don't think a mechanisms exists to address this type of functionality. I could be wrong.

Steve does have a spot in the Charge Point Details information for the end point address. Possibly this could be used for such a functionality?

2. The OCPP side of openevse would need to be able to handle having the websockets end point url updated via the backend after first initial connection.

Just looking for any thoughts about such a functionality or if this has been thought about in any way @matth-x .

Your time is appreciated! :smile:

jeremypoulter, 2024-08-07:
@matth-x any thoughts?

matth-x, 2024-08-08:
Hi @retrocoder-78, sorry, I must somehow missed this issue.

The current provisioning scheme is to define factory defaults and then dynamically configuring the connection details via OpenEVSE dashboard or the default OCPP server.

The factory defaults are hardcoded into the firmware here:

https://github.com/OpenEVSE/openevse_esp32_firmware/blob/fe5be4d600bb689010081315cc7244ef2cbea91f/src/app_config.cpp#L194-L196

If you set them to your server, OpenEVSE will initially connect to it. Using the ChangeConfiguration command, you can then update the following configs:

- Cst_BackendUrl: OCPP server "base URL"
- Cst_ChargeBoxId: identifier of the charger
- AuthorizationKey: Websocket key

Actually the custom Configs are implemented the same way as the standardized configs in OpenEVSE, so Change-/ and GetConfiguration work for the connection details configs too.

retrocoder-78, 2024-08-16:
Thank you Matthias! This makes sense. Is it at all possible that an
entry in Steve could adjust the WS (WSS) configuration in the OpenEVSE
dashboard? This is what in some way other OEM's have introduced into their
firmware and backend.

Steve does have a field labelled,

*Endpoint Address*. I believe this is the WS address entered into the
OpenEVSE dashboard like you mentioned.

From how I have read the OCPP standard, and looking through Steve, the
answer is it would be a custom feature on both sides.

The way I read your reply is, that currently, no. It's setup to enter into
the OpenEVSE firmware, and then communicate with Steve.

Any thoughts you have would be helpful! 🙂

matth-x, 2024-08-18:
Yes, it is possible. Sorry for the confusing explanation in my previous reply.

In SteVe, you can use the ChangeConfiguration operation to update the backend URL and chargeBoxId of the OpenEVSE. In the ChangeConfiguration view, select the key type "Custom", enter "Cst_BackendUrl" in the custom configuration key field and the server address in value. Press the perform button and then, do the same with custom configuration key "Cst_ChargeBoxId" and the new value you want to assign.

To streamline this process, you could define a factory default OCPP server URL which is hardcoded on the OpenEVSE. Then, each newly deployed OpenEVSE would first connect to the factory OCPP server where you can assign the final OCPP server address comfortably via ChangeConfiguration command.

retrocoder-78, 2024-08-20:
This is great information! Thank you for the detailed explanation. I gave this a try, changing the boxid via Steve to have a change in the OpenEVSE unit. What came back from Steve is, 'not supported'. I have that boxid setup in Steve.

Any thoughts?

[image]
[image]

closed 2026-09-21 by github-actions[bot]