OpenEVSE/openevse_esp32_firmware #886: TFT display doesn't show correct max or actual charging current

lutorm, 2024-07-22:
The old text display would show the actual current while charging. The new TFT display does not, but it would be nice if it did. The max charging current that it does show also does not update based on shaper settings, my station is set to 16A and even if the shaper cuts the available charge current to something less, the display still shows 16A. This seems less than ideal.

ausil, 2024-07-31:
Somewhat related, I have had the Amperage vanish entirely on both of my units.  I agree that it should show the current charge rate while charging.  and the maximum possible when not plugged in

ClassicGOD, 2024-08-07:
I'm having the same issue but I just assumed it's early days for the TFT and it's going to get better. 
I would love to see real time not only set current but real charging current, power ETC.

clowrey, 2024-08-15:
Hey @lutorm cool to see you on here! I just got my own OpenEVSE online (I've installed one 2 years ago at a friends) and thought the same thing - it seems this open source firmware for the screen etc is really just a beta type display of a few settings. Not sure where the active development discussions are taking place other than the issues here.

eagle360, 2024-11-05:
I don't know if this is related to this issue but I have now had a new OpenEVSE with the new 3.5" tft display freeze up twice in the last two weeks.  It has only been in use for the last two weeks and is running the configuration below as delivered.  The first time the display froze 17 minutes into a charge and the second time about 30 minutes.  In both cases the entire display and Web server were unresponsive and required a power cycle.  The OpenEVSE module was still going and the charge continued after the display module was locked up.  Apparently there is no watchdog timer enabled on this module that would reset it when it locks up.  While not ideal, as any logging data would be lost, it would at least keep the unit functional.  A boot notification would allow knowing that the system reset.  I will see if this issue continues to happen with the developer build and post in the issue thread if it continues to happen. Also as far as I know there were no web browsers connected to the device when it locked up.  The device is still a mostly default configuration with only two timers and an NTP server configured. The rest of the configuration is as delivered. 

OpenEVSE     8.2.2     
OpenEVSE Wifi     v5.1.2

closed 2026-09-21 by github-actions[bot]