OpenEVSE/openevse_esp32_firmware #892: Open EVSE V4.0 Wifi Issue

Nick425321, 2024-08-16:
I have two Open Evse's one is a v4.0 and the other is a v5.0
The v5.0 has 8.2.2 firmware and 5.1.1 wifi firmware.
The v4.0 has 8.2.2 firmware.

I had both units on wifi firmware 4.2.2 and updated to 5.1.1
The v5.0 works normally.
The v4.0 boots without initializing the wifi, if I move the ESP-32 from the 4.0 to 5.0 unit it works as expected.
I also tried pre-release 5.1.2, same issue.

I down graded the v4.0 ESP-32 to 4.2.2 so it is usable.

jeremypoulter, 2024-08-16:
Sorry, maybe I am missing something but why do you need to use v4 if v5 is working?

Nick425321, 2024-08-16:
I have two vehicles.

jeremypoulter, 2024-08-16:
but that still doesn't explain the need to keep using v4 both units should be able to run v5 with no issues

Nick425321, 2024-08-17:
I have one Open EVSE that has version 4.0 hardware (controller) on firmware 8.2.2 and it has an ESP32 on WiFi firmware 4.2.2. This is the problem unit.

I also have one Open EVSE that has version 5.0 hardware (controller) on firmware 8.2.2 and it has an ESP32 on WiFi firmware 5.1.2. This unit works as expected.

It is the older EVSE that will not work with any WiFi firmware newer than 4.2.2. 

I hope this makes sense, too many version 4's and 5's involved. My last sentence in the original post was clear to me, but now when I read it I realized it is poorly worded.

chris1howell, 2024-08-17:
There is very little difference from v4 hardware to v5 hardware, other than
connectors. The v4 hardware issue likely is something power related. There
is no initialization process, the WiFi will boot it it has power.

closed 2026-09-21 by github-actions[bot]