OpenEVSE/openevse_esp32_firmware #894: Hyundai Tucson PHEV Chaging Issues

blaxer, 2024-08-19:
Having some major issues getting a 2024 Hyundai Tucson PHEV to charge correctly on a brand new OpenEVSE charger. When the vehicle is initially plugged in, the vehicle status just starts bouncing like crazy (see below). Through trial and error the only way I can get it to charge is to reset the evse and then the esp32/wifi at which point it will start charging at 6a and then I have to drag the slider to any other value and then back to 48a in order to get it up to ~36amp charging. 

The real problem is that as soon as the charging has completed the status bouncing starts all over again, **this seems to be keeping the car "alive" which is killing the 12volt battery** and we arrive at a dead car in the morning. 

I am pretty confident the charger is working fine as I have a fully electric vehicle that charges with no issues at all on the same charger. I am on EVSE 8.2.2 and using the "modified" 5.x firnware for OpenEVSE Wifi. 

[image]

I would greatly appreciate any thoughts or fixes, I have read through a huge number of issues on here trying to make some sense of this but haven't found anything that seems to be the same issue.

TIA!

KipK, 2024-08-19:
Can you try to set the EVSE / "Pause mode" to disabled instead of sleep ?

blaxer, 2024-08-19:
I think I tried that, but I think I tried everything at one point.. but I will try it again and report back.

mr-rfh, 2024-08-22:
I have a very similar issue with status bouncing on my Chevrolet Volt. Versions same as blaxer's. But my status is bouncing one time per second. Here is some communication data from OpenEVSE console, maybe it could help?

`$FB 5^35
$OK^20
$FB 6^36
$OK^20
$FP 0 1 Time~~9:24:29~AM^E0
$OK^20
$SY^2E
$OK 5 6 0^33
$AT fe 03 16 0600^30
$FB 5^35
$AT fe 03 16 0300^35
$OK^20
$FP 0 1 Time~~9:24:29~AM^E0
$OK^20
$AT fe 03 16 0600^30
$FP 0 1 Time~~9:24:30~AM^E8
$AT fe 03 16 0300^35
$OK^20
$SY^2E
$AT fe 03 16 0600^30
$OK 5 6 0^33
$FP 0 1 Time~~9:24:30~AM^E8
$AT fe 03 16 0300^35
$OK^20
$FB 6^36
$OK^20
$FP 0 1 Time~~9:24:31~AM^E9
$OK^20
$SY^2E
$OK 5 6 0^33
$AT fe 03 16 0600^30
$FB 5^35
$AT fe 03 16 0300^35
$OK^20
$FP 0 1 Time~~9:24:32~AM^EA
$OK^20
$SY^2E
$AT fe 03 16 0600^30
$OK 5 6 0^33
$FP 0 1 Time~~9:24:32~AM^EA
$AT fe 03 16 0300^35
$OK^20
$FB 6^36
$OK^20
$FP 0 1 Date~~2024-08-22^2B
$OK^20
$AT fe 03 16 0600^30
$FB 5^35
$AT fe 03 16 0300^35
$OK^20
$SY^2E
$OK 5 6 0^33
$FB 6^36
$OK^20
$SY^2E
$AT fe 03 16 0600^30
$OK 5 6 0^33
$FB 5^35
$AT fe 03 16 0300^35
$OK^20
$FB 6^36
$AT fe 03 16 0600^30`

closed 2026-09-21 by github-actions[bot]