OpenEVSE/openevse_esp32_firmware #897: Problem Updating Firmware

Matol, 2024-08-21:
I can't install the latest firmware on the station. I open 'firmware update' window and try both options mentioned below:
1. automatic update via direct download from github - it says that update has started and then nothing happens.
2. manual update, I download the file from github (openevse_wifi_tft_v1.bin) and upload it to the station, it shows a progress bar and it stops near the end and again nothing happens. I reboot the station and it shows old firmware version.

Current version is: openevse wifi 5.1.0_modified, latest on github: 5.1.2

Any help or pointers will be greatly appreciated

glynhudson, 2024-08-21:
Can you try updating via the CURL method: https://openevse.stoplight.io/docs/openevse-wifi-v4/3e38d9ead39fc-user-guide#via-web-interface

Matol, 2024-08-22:
Thank you for the suggestion. Unfortunately it failed and we got an "Error: 4".
The update screen is attached hereto.
[image]
Any pointers or ideas will be great:)

ClassicGOD, 2024-08-24:
This is the same issue as here: https://github.com/OpenEVSE/openevse_esp32_firmware/issues/881
In short your options are: update manually using UART adapter or wait for official solution. The issue is probably the incorrect partition file flashed in the factory and probably all recent TFT boards are affected.

clowrey, 2024-08-25:
I had the same issue, I made my own uart adapter and then the bin file can be flashed easily even by web tools like https://web.esphome.io/ etc

mshoe007, 2024-09-07:
Just to add more info. When get the openevse_wifi_tft_v1.bin file from github and use the 'select file' method, I can see from the developer console in another browser window:

```
Update Start: openevse_wifi_tft_v1.bin 1995625
Update: 0%
...
Update: 98%
```

That's as far as it gets.

chris1howell, 2024-09-09:
The repartition process for the TFT LCD in the short term requires a USB => Serial adapter and esptools (requires Python and pyserial). 

Download the 5.1.2 - openevse_wifi_tft_v1.bin, bootloader_16mb.bin and partitions_16mb.bin from

https://github.com/OpenEVSE/openevse_esp32_firmware/releases/tag/v5.1.2

Connect the module to the USB => Serial adapter, then put the module in bootloader mode by holding BOOT and then pressing RESET then release BOOT

Run the command:

esptool --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_size=detect 0x1000 bootloader_16mb.bin 0x8000 partitions_16mb.bin 0x10000 openevse_wifi_tft_v1.bin

robao23, 2025-02-19:
[image]

I have the same problem here; it never updates.

markhellier, 2025-04-12:
I have 2 chargers on v5.1.0_modified is there a walk through for updating. I’ll need to buy a serial programmer.

olivervhansen, 2025-04-14:
@markhellier try the steps mentioned in https://github.com/OpenEVSE/openevse_esp32_firmware/issues/881#issuecomment-2694343259

closed 2026-09-21 by github-actions[bot]