OpenEVSE/openevse_esp32_firmware #900: PN532 with advanced series (large TFT)

johnvcoleman, 2024-08-30:
I’ve just installed a PN532 board, connecting it to the qwiic connector on the back of the TFT. The board powers up, but there doesn’t appear to be any communication. 

1. On power up the PN532 board gave a red status light.
2. I went into the UI and turned on the RFID setting. EVSE status lights turned purple.
3. Tapped the scan button, it began counting down, but no amount of card waving produced a read.

I tried several RFID cards and double checked my wiring, no joy.

Can anyone spare some troubleshooting tips or links to troubleshooting guides?

The PN532 board:
https://www.amazon.com/gp/product/B01I1J17LC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I Attached the data export and an image. Yes, I know, I was very careful taking the picture and had adult supervision. :)

Thank you!

[image]

[RFID-export.json](https://github.com/user-attachments/files/16822770/RFID-export.json)

clowrey, 2024-08-31:
Did you check the address switch on the board? It has to be in the right position to match whatever address is set in the firmware. I have that same module it works, but I forget what position the switches have to be in.

johnvcoleman, 2024-08-31:
The silkscreen next to the switches list the settings as HSU, I2C, and SPI. I have them set to I2C. (admittedly, that was a guess)

chris1howell, 2024-08-31:
I2C is correct. Make sure SCL and SDA are correct. The pin out on the
OpenEVSE side is standard Sparkfun Quiic.

clowrey, 2024-08-31:
Dont know then.. Its possible the RFID board is defective or something else, those cheap modules are rarely even tested after SMD assembly so some have faults.. But I cant say much else from here, better pictures might help ;) It does look like your wiring is correct.

clowrey, 2024-08-31:
Actually I think your DIP switch is in the wrong position, unless you adjusted it after the picture was taken. It should have one ON and one OFF
[image]
[image]

johnvcoleman, 2024-08-31:
Good eye, I spotted the switch settings after taking that photo. Sorry. I’ll double check the wires again. I guess it’s also possible I got a DOA board.

johnvcoleman, 2024-09-02:
Ok, got a new board. (At least Amazon is fast.) Same behavior, board powers up (using i2c switch settings) but does not appear to read or communicate.  Then I remembered that I've been running a dev build from March (I updated to the dev build to fix a TFT issue) Fearing that not being on the most recent firmware was causing the issue with the reader I've spent the last hour or so trying to update the firmware, which consistently hangs around 98% and never completes. I Reset the device, completed the wizard, attempted the firmware update again (both release and dev) and it just hangs. (I waited 15 minutes before giving up) It looks like I may need to get the serial programmer and do this the hard way.

[image]
[image]

clowrey, 2024-09-02:
Yes make a serial adapter like this. For the price the LCD board should definitely come with USB port IMO. 
[image]

chris1howell, 2024-09-09:
The partitions for the OS and Over the air update need to be increased in size. 

The repartition process for the TFT LCD in the short term requires a USB => Serial adapter and esptools (requires Python and pyserial). 

Download the 5.1.2 - openevse_wifi_tft_v1.bin, bootloader_16mb.bin and partitions_16mb.bin from

https://github.com/OpenEVSE/openevse_esp32_firmware/releases/tag/v5.1.2

Connect the module to the USB => Serial adapter, then put the module in bootloader mode by holding BOOT and then pressing RESET then release BOOT

Run the command:

esptool --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_size=detect 0x1000 bootloader_16mb.bin 0x8000 partitions_16mb.bin 0x10000 openevse_wifi_tft_v1.bin

rbray89, 2025-07-15:
[image]
[image]

Looks like your connections are fine. Make sure the DIP switch is set correctly. Note that the reader likes to read in the orientation I have mine in, with the traces towards the RFID chips. I'm running the latest build now.

closed 2026-09-21 by github-actions[bot]