OpenEVSE/openevse_esp32_firmware #901: Oscillating setpoint in Export mode when BOTH solar production and grid powers are published via MQTT

jamarju, 2024-08-31:
Hello, first and foremost thanks for this GREAT project, both the hardware and the software are fantastic!

It seems that there is an issue in the divert control system, where the current setpoint keeps going up and down without settling. This happens when both solar power production **AND** grid import/export values are reported through MQTT.

I'm latest firmware (5.2.0). My current set-up is as follows (screenshots below):

- Self production: enabled
- Mode: Export
- Feed: +Import/-Export power from Home Assistant

I was also experimenting with `Production` mode, which is set up this way:

- Mode: Production
- Feed: PV production feed from the inverters via Home Assistant

But note that my currently selected mode is: `Export`.

In this scenario, the setpoint oscillates. I have compiled the firmware in debug mode (target `openevse_wifi_tft_v1_dev`) with the `-D ENABLE_DEBUG_DIVERT` option enabled. See an example trace and walkthrough below:

```
{"amp":19360,"voltage":233.5,"power":4520.56}
voltage = 233.50
Igrid_ie = -12.87
amps = 19.36
Igrid_ie = -32.23
reserve = 1.28
_available_current = 30.94
_smoothed_available_current = 33.07
_charge_rate = 30
min_charge_time_remaining = 0
{"claims_version":1}
{"divert_update":0,"grid_ie":-3004,"divert_active":true,"charge_rate":30,"trigger_current":6,"voltage":233.5,"available_current":30.94029979,"smoothed_available_current":33.0734289,"pilot":21,"min_charge_end":0}
{"pilot":30}
{"amp":19360,"voltage":233.5,"power":4520.56}
{"evse_connected":1,"amp":19360,"voltage":233.5,"power":4520.56,"pilot":30,"max_current":33,"temp":355,"temp_max":355,"temp1":false,"temp2":355,"temp3":false,"temp4":false,"state":3,"status":"active","flags":1344,"vehicle":1,"colour":6,"manual_override":0,"freeram":113556,"divertmode":2,"srssi":-63,"time":"2024-08-31T11:32:40Z","local_time":"2024-08-31T13:32:40+0200","offset":"+0200","uptime":228263,"session_elapsed":2249,"session_energy":2766.183585,"total_energy":627.4020846,"total_day":2.766183585,"total_week":14.66208464,"total_month":14.66208464,"total_year":14.66208464,"total_switches":50,"elapsed":2249,"wattsec":9958260.906,"watthour":627402.0846}
{"amp":19360,"voltage":233.5,"power":4520.56}
{"session_elapsed":2251,"session_energy":2769.037816,"total_energy":627.4049388,"total_day":2.769037816,"total_week":14.66493888,"total_month":14.66493888,"total_year":14.66493888,"total_switches":50}
{"amp":21780,"voltage":233.5,"power":5085.63}
{"amp":21780,"voltage":233.5,"power":5085.63}
voltage = 233.50
Igrid_ie = -12.87
amps = 21.78
Igrid_ie = -34.65
reserve = 1.28
_available_current = 33.36
_smoothed_available_current = 33.09
_charge_rate = 33
min_charge_time_remaining = 0
{"claims_version":2}
{"divert_update":0,"grid_ie":-3004,"divert_active":true,"charge_rate":33,"trigger_current":6,"voltage":233.5,"available_current":33.36029979,"smoothed_available_current":33.09193007,"pilot":30,"min_charge_end":0}
{"pilot":33}
{"amp":25300,"voltage":233.5,"power":5907.55}
{"amp":25300,"voltage":233.5,"power":5907.55}
{"amp":28820,"voltage":233.5,"power":6729.47}
{"voltage":232}
voltage = 232.00
Igrid_ie = 7.02
amps = 28.82
Igrid_ie = -21.80
reserve = 1.29
_available_current = 20.51
_smoothed_available_current = 33.03
_charge_rate = 20
min_charge_time_remaining = 0
{"claims_version":3}
{"divert_update":0,"grid_ie":1628,"divert_active":true,"charge_rate":20,"trigger_current":6,"voltage":232,"available_current":20.50965517,"smoothed_available_current":33.02917571,"pilot":33,"min_charge_end":0}
_pause_timer = 0
{"pilot":20}
{"session_elapsed":2257,"session_energy":2778.093795,"total_energy":627.4139948,"total_day":2.778093795,"total_week":14.67399486,"total_month":14.67399486,"total_year":14.67399486,"total_switches":50}
{"amp":25960,"voltage":232,"power":6022.72}
{"amp":25960,"voltage":232,"power":6022.72}
{"amp":18260,"voltage":232,"power":4236.32}
```

Walking through the trace:

1. The trace starts out while the car it's charging at 19.36A
2. A new +I/-E report comes via MQTT: -12.87A
3. Car amps are subtracted from grid amps (-12.87 - 19.36 = -32.23), 1.28 amps are reserved, 30.94 available, thus new setpoint is: 30. So far so good.
4. A new PV production report comes via MQTT, doesn't matter its value as it's not used because we are in `Export` mode. However, and I believe here is where the bug lies, [the setpoint calculation code is called again here](https://github.com/OpenEVSE/openevse_esp32_firmware/blob/878c400181aa63efe744be1fc9aa59a46cd9afe4/src/mqtt.cpp#L72).
5. Car amps (21.78), which are on its way to ramping up to their current setpoint (30), are subtracted from the OLD grid amps, which hasn't updated yet (-12.87 - 21.78 = -34.65), 1.28 amps are reserved, 33.36 are available and new setpoint is 33.
6. A new +I/-E report comes via MQTT: +7.02A as the previous setpoint overshot the actual available current.
7. Car amps are subtracted from grid amps (7.02 - 28.82 = -21.8), 1.29 are reserved, 20.51 available thus the new setpoint is 20A.

This over-shooting / under-shooting the setpoint (30 -> 33 -> 20) keeps going on forever.

If I stop reporting PV production from Home Assistant, the issue resolves.

However, that would prevent me from using the `Production` mode.

I believe the solution should be to not call `divert.update_state();` from MQTT callback if `topic_string == mqtt_solar` and we're in `Export` mode but I'm not aware of other implications.

I can raise a PR with a suggested solution for review.

Thanks again!

### Export mode config screenshot

[image]

### Production mode config screenshot

[image]

jeremypoulter, 2024-08-31:
Hum... You shouldn't be able to setup to use both Grid IE and Solar at the same time. They should be mutually exclusive, so if that is happening then the bug is that we are allowing both to be selected. Can you describe how you set these up?

jamarju, 2024-08-31:
Hi Jeremy. I just set it up the way it looks in the screenshots.

Then, on the Home Assistant side, I have 3 automations that publish the following topics:

- PV power: `ha/sensor/inverters_input_power`
- Grid +I/-E: `ha/sensor/power_meter_active_power_pos_import`
- Voltage: `ha/sensor/power_meter_voltage`

You are correct in that the two modes are mutually exclusive (ie. only one can be selected in the combobox), but the problem is that both `mqtt_solar` and `mqtt_grid_ie` remain set to their respective topics, thus on [mqttmsg_callback](https://github.com/OpenEVSE/openevse_esp32_firmware/blob/878c400181aa63efe744be1fc9aa59a46cd9afe4/src/mqtt.cpp#L57) `divert.update_state()` gets called twice.

jeremypoulter, 2024-08-31:
Does a reboot 'fix' the issue?

jeremypoulter, 2024-09-01:
I think this is an issue with not unsubscribing from one topic when you change the config: https://github.com/OpenEVSE/openevse_esp32_firmware/blob/master/src/mqtt.cpp#L381-L390 so if you reboot it should work ok, even if both are configured

lutorm, 2024-10-15:
I'm also occasionally seeing the shaper setpoint oscillate (without solar divert, just normal shaper). As was alluded to above, it seems to happen because the reported grid power comes in late, and the shaper code subtracts off the instantaneous charge current. If the charge current has ramped up but is not yet reflected in the power value, the shaper will calculate a too large available current. Once the charge current has ramped up to this value and it's reflected in the power, the shaper cuts back sharply. This causes the charge current to drop but until this is reflected in the power value, an abnormally low available current is calculated and this usually causes the charge to stop. Then the cycle repeats, leading to repeated current overshoots and relay chatter.

It's generally iffy to construct differences of quantities that aren't measured in the same timebase, like the internal charge current and the live power, since transients will lead to abnormal values. I'd suggest we put a filter on the shaper output _maxpwr to damp these transients. A simple IIR filter should suffice. Yes, putting a filter on it will slow down the ramp down when max power is exceeded, but a single short exceedance seems preferred over the repeated relay chatter that can happen now.

kwauchope, 2025-01-25:
Just to add another datapoint in case it is of use. I've seen this issue as well, large overshoot followed by large undershoot causing charge to stop while using export mode. In my loop there is quite a delay from the raw export value sample and it being posted to the MQTT topic, average delay is > 10s.

As a fix I used the simple patch posted by @goldserve in [#685](https://github.com/OpenEVSE/openevse_esp32_firmware/issues/685#issuecomment-1615124001)

closed 2026-09-21 by github-actions[bot]