OpenEVSE/openevse_esp32_firmware #905: HTTP API request /ws not working

dmitry-zakharov-qs, 2024-09-03:
Hello.

I'm running latest 5.1.2 version of firmware. I want to track EVSE status in real time via HTTP API.
I can get initial status with `/status` request as per
https://openevse.stoplight.io/docs/openevse-wifi-v4/dba95e8966593-get-the-evse-status
Then, as advised there, I try to use `/ws` request for status updates but not getting any when a car is connected and charging starts and stops. I tried to open `/ws` in browser and with curl:
`curl --request GET --url http://10.1.30.39:80/ws --header 'Accept: application/json'`
with the same result - no updates.
I'm using EV simulator for my testing if it matters.

Any help is greatly appreciated.
Thanks.

KipK, 2024-09-03:
It is a websocket, , it streams data over it. You can't poll it using http get. 

https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API

KipK, 2024-09-03:
You can however test it using curl like this:

```
curl --include \
     --no-buffer \
     --header "Connection: Upgrade" \
     --header "Upgrade: websocket" \
     --header "Host: openevse:80" \
     --header "Origin: http://openevse:80" \
     --header "Sec-WebSocket-Key: SGVsbG8sIHdvcmxkIQ==" \
     --header "Sec-WebSocket-Version: 13" \
     "http://openevse:80/ws"
```

However it's better to use some dedicated tools for wss:// protocol like wscat

dmitry-zakharov-qs, 2024-09-04:
Thanks so much for you help. It does work via websocket. Even curl can read it simply like
`curl ws://10.1.30.39:80/ws`
But it seems then the sample request at
https://openevse.stoplight.io/docs/openevse-wifi-v4/7efd14b814ec2-evse-status-updates
should be corrected.

closed 2024-09-04 by dmitry-zakharov-qs