OpenEVSE/openevse_esp32_firmware #928: Can not calibrate current scale or offset

chris1howell, 2024-11-04:
[Patrik Jonsson] Posted on OpenEVSE Forum...

(https://openev.freshdesk.com/users/6059459827)

I want to fix the current sensor calibration on my unit and the instructions I've found say to do this with RAPI commands, which doesn't seem to work (I can see the value but not alter it using the dev console.) Are there updated instructions for how to update the calibration without RAPI if this is no longer available? I don't see the values in the web configuration anywhere.

The RAPI commands for scale and offset are blocked but the http interface does not have the calibration UI.

lutorm, 2024-11-06:
I realized I can update the calibration by downloading the settings json and editing it before uploading it back. Not super user friendly, but given this only needs to be done at most once it's not a big deal.

KipK, 2024-12-01:
Hi,

I'm currently adding this to UI.
However seems sending "scale" and "offset" value to /config doesn't works here. It still sends back default values after update.

@lutorm, do you confirm after POST the config data to /config endpoint you got scale and offset value updated thereafter ? 

It's not working for me here, need to find why

lutorm, 2024-12-01:
It did work, but I might have needed to reboot before it showed up. I seem
to remember there was some weirdness.

KipK, 2024-12-01:
You're right. Values are updated on EVSE module, but it seems the config endpoint doesn't get latest value after updating it. 

I'll fix that. thanks

KipK, 2024-12-01:
For now, I've added it in EVSE tab
[image]

I'll think about what would be the best way to add this calc tool ( a new Tools sections, or a popup from the EVSE / current sensor section )

closed 2026-09-21 by github-actions[bot]