OpenEVSE/openevse_esp32_firmware #930: WiFi unresponsive - possible memory leak

joeyberkovitz, 2024-11-15:
After leaving the openevse running for a while - (often months), the Wi-Fi interface becomes very slow

Page resource loading takes effectively infinite time, although an API request to trigger a reboot does work a bit slowly.

I suspect a slow memory or resource leak somewhere. Is there any way to debug this, or even just add some sort of watchdog that reboots the wifi module in a case like this?

njh, 2024-12-18:
Which browser / Operating System are you using?
What is your Wifi firmware version?

joeyberkovitz, 2024-12-18:
I've observed this with the current firmware on a pre-TFT model, and with the TFT on 745/merge_52e4a6ff_modified (currently stuck there due to the incorrect partitioning on early models) 

Browser is always chrome, on windows/Linux/android

To be clear, when the issue occurs, a standard curl to load any of the HTML/JS/CSS resources is extremely slow to the point that it doesn't work in the browser. A REST reboot via curl or similar still works slowly and then the webpage loads

njh, 2024-12-18:
Sorry, the curl problem might be a red herring.

This takes ~5.8 secs but works:
```
curl -sv 'http://openevse.local/' | gunzip -
```
I suspect it shouldn't be returning `Content-Encoding: gzip` without the client providing `Accept-Encoding` in the request.

joeyberkovitz, 2024-12-18:
On my TFT model, currently functional, that curl takes 0.03 seconds

When it's non functional, that's significantly slower.

joeyberkovitz, 2024-12-26:
So I did some more testing:

- if you go to `http://openevse/status`, there's a `free_heap` and `freeram` field returned. For example, it starts at roughly 77968
- if I refresh the status page a few times, the memory usage doesn't change much, fluctuating up/down a few bytes
- if instead I go to `http://openevse`, loading all of the resources (websocket, HTML, ...), then close that tab, it drops to 76264 - so roughly 1704 bytes were used and never returned

It's hard to debug, but if I had to guess, the webserver generically, or the websocket more specifically is either leaking memory on every page load, or just not detecting closed connections well.

I think an easy fix might be a watchdog, or similar, that triggers an ESP32 reboot if memory usage drops below a percentage threshold. A more sustainable, but probably more complex fix would involve finding the source of the leak, which may lie in imported libraries.

Please let me know if there's any info I can provide.

njh, 2024-12-26:
Hm, looks like I am experiencing an unrelated problem.
I will open a new issue.

If I make the request to `/` 20 times,  I do not see the available memory consistently dropping - it goes up and down.

```sh
for i in $(seq 1 20);
do
    curl -s -o /dev/null 'http://openevse.local/' 
    curl -s 'http://openevse.local/status' | jq '.free_heap'
done
```

Result:
```
194988
194936
194924
194940
194912
194956
194920
194968
194944
194940
194928
194936
194916
194964
194940
194936
194968
194944
194944
194932
```

(Running Wifi Gateway version v5.0.2)

joeyberkovitz, 2024-12-26:
Curl to / isn't really sufficient because it doesn't load all of the resources or open the web socket like the browser does.

njh, 2024-12-26:
Ah, yes, of course. Good point.

```sh
for i in $(seq 1 50);
do
    #curl -s -o /dev/null 'http://openevse.local/'
    open -a Firefox -u 'http://openevse.local/'
    sleep 6
    curl -s 'http://openevse.local/status' | jq '.free_heap'
    osascript -e 'tell application "Firefox" to quit'
    sleep 2
done
```
(quitting Firefox is overkill but it is easy/works)

And the Free RAM results look like this:
[image]

(starts at 193.7k and finishes at 180k)

So just need to work out which HTTP call is leaking...

closed 2026-09-21 by github-actions[bot]