OpenEVSE/openevse_esp32_firmware #933: OpenEVSE initiatiates charge upon startup

brian-mcnamara, 2024-11-26:
It seems like no matter what I try, the behavior when a vehical is connected, that it will go into charging state when device is powered.

I have tried a bunch of things, my setup is to use divert/eco mode. I do see the eco mode claim to disable charge is created at startup, however the contactor engages seconds after powering the device when the veichial is connected. I tried disabling that, and setting default state to disabled, as well, however the same issue happens.

There seems to be some default behavior when the vehical is connected, that openevse will go into charging state upon startup. This conflicts with the default claim devert applies.

chris1howell, 2024-11-27:
In order to give full control of the OpenEVSE controller you need firmware
8.2.3 with heartbeat supervision. The controller will boot locked and
remain so until unlocked by the ESP32. The upside is the ESP32 has full
control. The downside is the ESP32 has full control, the station will not
operate as a dumb station and any issue with the ESP32 means the station is
a brick.

brian-mcnamara, 2024-11-27:
Thanks @chris1howell . I literally found the same thing as I was navigating the openevse source code and finding the BOOTLOCK. I just updated the openevse base board off master and that did it! I'm willing to rely on the ESP32 for the total control it can provide. Worse case, I know how to flash :).

Thanks for chiming in so quickly!

closed 2024-11-27 by brian-mcnamara