OpenEVSE/openevse_esp32_firmware #943: wrong screen color

monfiston, 2024-12-26:
hello
i have 2 openEVSE, one official and one tinkered by myself
the color of the screen is not identical on the 2 openevse,
is there a way to change the color of the screen?

[image]

jeremypoulter, 2024-12-26:
Is the LCD an OpenEVSE LCD on both?

monfiston, 2024-12-26:
No, it's not an original screen, but I thought that in the past you could choose a specific color?

jeremypoulter, 2024-12-26:
The LCD colour is linked to the EVSE state. As it is lighting up and displaying text it must be the same driver chip, but I assume the RGB backlight is connected to the driver in a different way (I think the red and green are swapped). So for proper operation you will need to do a custom version of the OpenEVSE that corrects for this.

monfiston, 2025-01-19:
hello
it is not possible to change the connection to the rgb led?
if not, what should I change in the code?

closed 2025-08-27 by jeremypoulter