OpenEVSE/openevse_esp32_firmware #949: Retire Proof of Concept and develop LCD display graphics.

chris1howell, 2025-01-20:
This is an example from SquareLine Studio GUI tool. I think it is a good reference and discussion topic.

[image]

chris1howell, 2025-01-20:
Here are the screens I think we need.

-Boot Logo
  -WiFi Portal SSID and Password QR code 
  -WiFi Connected IP address
-Home Screen
  -Connected Screen
  -Auth Screen Scan RFID card or QR (station ID) if RFID and/or OCPP enabled
  -Waiting for Timer 
-Charging
-Connected Charging ended Session Stats 
-Error Screen

TommySharpNZ, 2025-01-30:
I’m assuming there’d also need to be a touchscreen to accomodate these mockups? Or at least some physical buttons…

closed 2026-09-21 by github-actions[bot]