OpenEVSE/openevse_esp32_firmware #952: History charge is innacurate

lebsd, 2025-02-15:
Set a schedule on 2/14 to start charging the car at 12:30 AM.  This morning to my surprise looking at my inverter data I find that the car charge started at 7:30PM (this may have been my mistake perhaps pressing the charge button vs leaving it in ECO mode).  However, when I look at the history, although I see an entry 2/14 at 7:26PM for charging, it only reflects 1.6KW when the inverter indicates up to 6KW from the grid. 

[image]

ClassicGOD, 2025-02-20:
I think you are reading the history incorrectly. 
kWh are not the same as kW. kW is a unit of power, kWh is a unit of energy. The 1.6kWh you see is the amount of energy used by the session up to this time.
What you see from your history is that the car was connected and the session meter was on 1.6kWh, then the charging started (again the meter stil on 1.6kWh) at 7:35PM and stopped at 9:22PM with the meter showing 18.9kWh. That means that it that time car used 16.4kWh of energy in that time, so was charging for 1h 47min with average power over that time of 9,2kW.

lebsd, 2025-02-22:
Thank you for your explanation, I did not realize that it was KWh.  Is
there a way to show the kW history  (besides just the statistics of per
day/wk/month/yr?

ClassicGOD, 2025-02-22:
> Thank you for your explanation, I did not realize that it was KWh.  Is
> there a way to show the kW history  (besides just the statistics of per
> day/wk/month/yr?

I don't think that there is a way to do that in the OpenEVSE itself. I don't think the microcontroller used to drive OpenEVSE has enough storage to store momentary usage. But there are multiple ways to do this using external services like EmonCMS - (official guide: https://openevse.dozuki.com/Guide/Services+-+EmonCMS/14?lang=en ) or by integrating with some smart-home software. For example I integrate mine with HomeAssistant.

closed 2026-09-21 by github-actions[bot]