OpenEVSE/openevse_esp32_firmware #953: Charging at 6A instead of 40A

1fish2, 2025-02-19:
The OpenEVSE was stuck at 6A (or 6.2A) charging rather than using the full 40A capacity. (See the sequence, below.)

(This seems unrelated to #812 because it occurred with and without any OpenEVSE timers, and in both cases with no other charge-limiting features besides system max 40A.)

**Hardware:** Replacement Electronics circuit board for Juicebox v2
**Firmware:** OpenEVSE 8.2.3.CGMI, OpenEVSE Wifi v5.1.2
**Connections:** On Wifi. No monitoring. No MQTT. No OCPP. No OhmConnect. No Home Assistant. No RFID.
**Settings:** No charge limits. Default state = Active, Three-phase = No, Service Level = 2, Pause Status = off, HTTP authentication = Disabled, Shaper = Disabled.

First day sequence (to the best of my memory):
1. Install the OpenEVSE circuit board. -> Charge rate defaults to 40A since that's the hardware's capacity and it's correct for the 50A circuit breaker.
2. Set up Wifi and time zone.
3. Set timers to disable charging during 4-9pm peak TOU hours.
4. Charge the Tesla. The Tesla is also set to wait out 4-9pm and it always has its own SoC charge limit. -> It started charging at 40A.
5. -> Charging paused at 4pm.
6. -> At ≈8:58pm the EVSE clicked and its LED switched from Cyan to Yellow.
7. -> At 9pm charging resumed but **at 6A** instead of 40A. The OpenEVSE and the Tesla agree on the actual charging current.
8. Toggle OpenEVSE to Disabled, then toggle it back to Auto. -> That didn't fix the charging rate. It finished charging overnight at 6A.

Second day sequence/experiment:
1. Delete the OpenEVSE charge timers.
2. Set Tesla to charge during 8pm - 4pm for this test, and with its own SoC max, as always.
3. Plug in the Tesla during 8-4pm. -> It waited until 8pm, then started charging at 6A.
4. Unplug and replug in the Tesla. -> Again charging at 6A.
5. The Tesla UI could pick either 5A or 6A charging current. -> The OpenEVSE must be signaling a 6A limit.
6. Disable Charge.
7. Restart OpenEVSE firmware. -> Charging at 6A.
8. Restart OpenEVSE Wifi firmware. -> Charging at 6A.
9. Reset OpenEVSE (on the Firmware screen).
10. Reconfigure OpenEVSE Wifi and time zone. -> Charging at 6A.
11. Disable Charge, then toggle it back to AUTO. -> Charging at 40A!!

1fish2, 2025-02-19:
Is it known what gets the OpenEVSE into 6A mode?
Is there a reliable way to get the OpenEVSE out of 6A mode?

danpadure, 2025-03-14:
I experienced something similar.
when charging at 6A I just red /claims and /override, without changing anything and this apparently "fixed" as charging speed increased

NEVdataDyne, 2025-03-19:
I am experiencing something similar : It always charges at 5,7A except when I change the max current or the service level, then the charging current increases to the max value and quickly drops back to 5,7A within a couple of seconds.
To my memory it always behaved like this (for months) but now that my solar panels are harvesting more energy, it has become an issue because my home battery gets full.
The device is controlled by Home assistant to manually set the charging current through the override action.

Firmware: OpenEVSE 8.2.2, OpenEVSE Wifi v5.1.2
Connections: On Wifi. No monitoring. No MQTT. No OCPP. No OhmConnect. connected to Home Assistant. No RFID.
Settings: No charge limits. Default state = Active, Three-phase = No, Service Level = any, Pause Status = off, HTTP authentication = Enabled, Shaper = Disabled.

1fish2, 2025-04-06:
@danpadure did you enter `/claims` into the Configuration > Developer > RAPI Commands (after the `$`), then hit `Send`, and get back `< $NK^21`? Ditto with `/override`? It's not working.

[image]

1fish2, 2025-04-06:
Charging amps are oscillating up and down between 5.94A to 15.84A while the Max amps are going back and forth between 6A vs. 40A.

Sometimes the car says it's drawing "8/40A", that is 8A from the then-allowed 6A!

**Q.** Do these Console logs and screenshots reveal what's happening?

These logs and screenshots were captured hours after I tried lots of ways to get it unstuck:
* disabling & re-enabling charging
* stopping & restarting charging from the car
* unplugging & replugging in the car
* setting Configuration > EVSE > Service Level to Auto or Level 1 then back to Level 2
* adjusting Configuration > EVSE > Max current down then back up to 40A
* adjusting max current in the car down and up (within the EVSE's range)
* turning a Charge Limit on then off
* sending the RAPI commands that @danpadure recommended

The OpenEVSE is configured with OhmConnect, unlike when I opened this Issue. It still has no timers, eco mode, monitoring, MQTT, OCPP, Home Assistant, RFID, HTTP authentication, or Shaper.

**OpenEVSE Console**

```
1 Manual�Override^E3
$AT 03 03 40 0540^34
$OK^20
$FP 15 1 ������^27
$OK^20
$SY^2E
$GG^24
$OK 8140 240000^2B
$GC^20
$OK 6 80 40 40^1E
$FP 0 0 Charging���5.94A^84
$OK^20
$FP 0 1 Manual�Override^E3
$OK^20
$FP 15 1 ������^27
$OK^20
$SY^2E
$GG^24
$OK 15180 240000^1B
$GC^20
$OK 6 80 40 40^1E
$FP 0 0 Charging���8.14A^81
$OK^20
$FP 0 1 Manual�Override^E3
$OK^20
$FP 15 1 ������^27
$OK^20
$SY^2E
$AT 03 03 6 0540^06
$GG^24
$OK 6160 240000^27
$GC^20
$OK 6 80 6 40^2C
$FP 0 0 Charging��15.18A^4F
$OK^20
$FP 0 1 Manual�Override^E3
$OK^20
$FP 15 1 ������^27
$OK^20
$SY^2E
$GS^30
$OK 03 44406 03 0540^13
$GG^24
$OK 5940 240000^2E
$GP^33
$OK -2560 186 -2560^3F
$GC^20
$FP 0 0 Charging���6.16A^8D
$OK^20
[... 332 lines cut ...]
Heartbeat failed
{"amp":5940,"voltage":240,"power":1425.6}
Heartbeat failed
{"amp":5940,"voltage":240,"power":1425.6}
{"session_elapsed":43569,"session_energy":23803.29695,"total_energy":317.2571392,"total_day":19.02793798,"total_week":19.02793798,"total_month":34.01987487,"total_year":87.35876693,"total_switches":56}
Heartbeat failed
{"amp":5940,"voltage":240,"power":1425.6}
Heartbeat failed
{"amp":5940,"voltage":240,"power":1425.6}
{"amp":5940,"voltage":240,"power":1425.6}
{"session_elapsed":43575,"session_energy":23805.68324,"total_energy":317.2595255,"total_day":19.03032428,"total_week":19.03032428,"total_month":34.02226116,"total_year":87.36115323,"total_switches":56}
{"amp":6600,"voltage":240,"power":1584}
Heartbeat failed
{"amp":13200,"voltage":240,"power":3168}
Heartbeat failed
{"amp":16280,"voltage":240,"power":3907.2}
{"session_elapsed":43581,"session_energy":23810.51336,"total_energy":317.2643556,"total_day":19.03515439,"total_week":19.03515439,"total_month":34.02709128,"total_year":87.36598334,"total_switches":56}
Heartbeat failed
{"amp":10340,"voltage":240,"power":2481.6}
{"evse_connected":1,"amp":10340,"voltage":240,"power":2481.6,"pilot":40,"max_current":40,"temp":183,"temp_max":233,"temp1":false,"temp2":183,"temp3":false,"temp4":false,"state":3,"status":"active","flags":1344,"vehicle":1,"colour":6,"manual_override":0,"freeram":166748,"divertmode":1,"srssi":-22,"time":"2025-04-06T16:38:22Z","local_time":"2025-04-06T09:38:22-0700","offset":"-0700","uptime":926537,"session_elapsed":43583,"session_energy":23811.98095,"total_energy":317.2658232,"total_day":19.03662199,"total_week":19.03662199,"total_month":34.02855887,"total_year":87.36745093,"total_switches":56,"elapsed":43583,"wattsec":8.572313142e7,"watthour":317265.8232}
Heartbeat failed
{"amp":5940,"voltage":240,"power":1425.6}
ohm_url = https://login.ohmconnect.com/verify-ohm-hour/Qy8KXeV2
{"session_elapsed":43588,"session_energy":23813.59188,"total_energy":317.2674341,"total_day":19.03823291,"total_week":19.03823291,"total_month":34.03016979,"total_year":87.36906186,"total_switches":56}
```

1fish2, 2025-04-06:
Screen recordings:

https://github.com/user-attachments/assets/3a5e0d06-f65f-41ee-aa34-4d8508c70aa1

https://github.com/user-attachments/assets/bf4804bd-6343-4a1c-821b-b2db27c3e1de

https://github.com/user-attachments/assets/1512a1a8-8b76-4ca3-a9d3-42b0b7f0dab6

KipK, 2025-04-07:
Seeing those "heartbeat failed" message, means heartbeat has been enabled and  the openevse module haven't received an heartbeat update so it caps the ampacity to 6A. 

I don't know what this heartbeat thing is used for, but seeing in the code you can disable it by RAPI by sending "SY 0"

Perhaps you've enabled it by mistake previously by sending an SY command?

1fish2, 2025-04-07:
@KipK interesting! I haven't issued any RAPI commands except for misunderstanding the web API commands that @danpadure recommended.

What I did that could be relevant was plug in the EV, then open the web API and turn off the charge energy session-limit that was set for charging the other EV. (It'd be useful to have a one-session charge energy limit.)

Perhaps using the web UI to turn off the charge energy session limit stopped the heartbeat sender but not the heartbeat listener?

(https://openevse.stoplight.io/docs/openevse-wifi-v4/ZG9jOjI5NzAxMTA1-rapi-api says it's no longer recommended to use RAPI set-commands since that can make the state conflict with the WiFi module, but the Web API doesn't appear to have heartbeat commands.)

[The RAPI SY info you quoted](https://github.com/openenergymonitor/open_evse/blob/master/firmware/open_evse/rapi_proc.h#L207-L215).

The EVSE is still logging "heartbeat failed" messages, 2 days later.

OpenEVSE Console is logging commands like these:
```
$GP^33
$OK -2560 224 -2560^34
$SY 165^3C
$SY^2E
$SY^2E
$SY^2E
$SY^2E
$SY^2E
$NK 5 6 2^30
$SY 165^3C
$OK 5 6 1^32
$AT 01 01 40 0200^37
```

Sampling the heartbeat status with a RAPI command:
```
> $GY
< $OK 5 6 1^32
```

Using the Firmware page of the web UI to Restart OpenEVSE did not stop the "Heartbeat failed" messages nor `$SY^2E` RAPI commands in the OpenEVSE Console and their `$OK 5 6 0^33` replies.

Using the Firmware page to Restart OpenEVSE Wifi **did** stop the "Heartbeat failed" messages, but **not** `$SY^2E` -> `$OK 5 6 0^33` in the OpenEVSE Console.

1fish2, 2025-04-08:
Testing charging after the Restart steps via the web UI, it's steady but slow at 6.2A, `{"amp":6160,"voltage":240,"power":1478.4}`. No "heartbeat failed" messages.

```
$FP 0 0 Charging~~~6.16A^8D
$OK^20
$SY^2E
$OK 5 6 0^33
$GG^24
$OK 6160 240000^27
$GC^20
$OK 6 80 40 40^1E
```

I could try issuing a `$SY` RAPI command with more info about what 0-2 parameters to use. rapi_proc.h and rapi_proc.cpp have clues. If the `$SY^2E` console lines are heartbeat messages, then they're arriving OK.

**Experiment**
* The EVSE was stuck at 6.2A. The Reset steps (above) seemed to end the "heartbeat failed" messages.
* Plug in the EV then set the per-session energy limit to 1kWh. -> It charged at 6.2A for about 40 min, adding 1kWh as expected. The idea was to hit the session energy limit so the code might clean up the state.
* Unplug the EV, delete the 1kWh limit, then plug in the EV (starting a new session).  -> It started charging at ≈6A again, but _this time_ the Tesla app showed `[ 6A > ]` meaning it could adjust its own limit upward from there, _unlike in yesterday's screenshot._
* Adjust Tesla's current up to 40A. -> It finally got back up to 10kW!

These results might point to the cause of the problem and to a potentially-repeatable workaround.

danpadure, 2025-04-08:
I haven't had any problems since last reporting, but I dont think I did anything to fix it....

KipK, 2025-04-08:
@jeremypoulter , I don't know how the heartbeat thing is handled. Do you have any clues on why it fails here? I don't have this problem on mine.

1fish2, 2025-04-08:
I tried these steps and some variations but didn't succeed in reproducing the problem:

1. Plug in an EV.
2. Use the web UI to set a session energy limit to 1 kWh.
3. Wait for it to charge up to the energy limit.
4. Unplug.
5. Plug in an EV.
6. Use the web UI to delete the session energy limit.

> Perform action 'OpenEVSE: Set override'

Would that be the same as toggling charge from AUTO to Disabled then to Enabled in the web UI?

Or POSTing a web request to http://openevse-xxxx.local/override with `--data '{"state": "disabled"}'` then another request with `--data '{"state": "enabled", "charge_current": 40, "max_current": 40}'`?

jeremypoulter, 2025-04-09:
> [@jeremypoulter](https://github.com/jeremypoulter) , I don't know how the heartbeat thing is handled. Do you have any clues on why it fails here? I don't have this problem on mine.

The main heartbeat code handler is here: https://github.com/jeremypoulter/OpenEVSE_Lib/blob/master/src/openevse.cpp#L897-L932

Within the OpenEVSE firmware this calls that method with the ack_missed with a value of true so missed heartbeats should be automatically acked and normal charging should recover, and you can see this happening:

```
$GP^33
$OK -2560 224 -2560^34
$SY 165^3C
$SY^2E
$SY^2E
$SY^2E
$SY^2E
$SY^2E
$NK 5 6 2^30
$SY 165^3C
$OK 5 6 1^32
$AT 01 01 40 0200^37
```

So heartbeat missed
```
$SY^2E
$NK 5 6 2^30
```

Send the recovery

```
$SY 165^3C
$OK 5 6 1^32
```

Current is returned to 40A:

```
$AT 01 01 40 0200^37
```

KipK, 2025-04-18:
And do you have an idea on why it miss this much pulses on their config? I never got that problem here.

1fish2, 2025-07-09:
I haven't seen this symptom repeat while using this charging procedure for 5 months:
1. Clear any charge limit set by partner.
2. Plug in my EV.
3. Charge with my EV controlling time-of-day and SoC limits.

That gives some evidence to narrow down the bug. I did try to find a way to reproduce it.

Q. Is there an open Issue on the other problems caused by a leftover charge energy limit?

It'd be useful to have the charge energy limit expire after the current charging session or else after reaching that limit, or to add a separate pop-up menu choice for single-session charge energy limit. Users always have to calculate the limit value for each session using the present and target SoC percentages so it's not really useful leftover for following sessions.

jeremypoulter, 2025-07-12:
There are two modes for the limits, a temporary limit that is cleared at the end of a session (set from the charge screen) or a limit that is applied to every session (configured in the settings, and should be reset each session)

danpadure, 2025-07-13:
1fish2: don't mix power with energy!
6-40A measures power, and that's the limits that jeremypoulter is talking about
what you mentioned above measures energy (kwh)

one of the limit must be evse max current from settings

danpadure, 2025-07-17:
ok, but this topic was about power limit, not energy

the0duke0, 2026-01-28:
> > Seeing those "heartbeat failed" message, means heartbeat has been enabled and the openevse module haven't received an heartbeat update so it caps the ampacity to 6A.
> > I don't know what this heartbeat thing is used for, but seeing in the code you can disable it by RAPI by sending "SY 0 6"
> > Perhaps you've enabled it by mistake previously by sending an SY command?
> > ```
> > SY heartbeatinterval hearbeatcurrentlimit
> >  Response includes heartbeatinterval hearbeatcurrentlimit hearbeattrigger
> >  hearbeattrigger: 0 - There has never been a missed pulse, 
> >  2 - there is a missed pulse, and HS is still in current limit
> >  1 - There was a missed pulse once, but it has since been acknowledged. Ampacity has been successfully restored to max permitted 
> >  $SY 100 6  //If no pulse for 100 seconds, set EVE ampacity limit to 6A until missed pulse is acknowledged
> >  $SY        //This is a heartbeat supervision pulse.  Need one every heartbeatinterval seconds.
> >  $SY 165    //This is an acknowledgement of a missed pulse.  Magic Cookie = 165 (=0XA5)
> >  When you send a pulse, an NK response indicates that a previous pulse was missed and has not yet been acked
> > ```
> 
> I have a JuiceBox EVSE that I just converted to OpenEVSE and running the SY 0 6 command solved my issue

I also just updated my JucieBox Pro 40 (v1) with the OpenEVSE board. Charging was essentially stuck at 6A despite setting the max rate to 40A. This was on OpenEVSE v8.2.3.CGMI and OpenEVSE Wifi v5.1.5. Issuing the `SY 0 0` command instantly resulted in the charge rate jumping to ~40A.