OpenEVSE/openevse_esp32_firmware #963: Safari issue with 304 responses

nathanejohnson, 2025-04-27:
According to the rfc: https://datatracker.ietf.org/doc/html/rfc7232#section-4.1 

   A 304 response cannot contain a message-body; it is always terminated
   by the first empty line after the header fields.

It looks like Mongoose is sending back a body on 304.

[image]

And this is the error message I see in Safari:

[image]

I was able to reproduce this issue putting together a little test program that would listen on 80 and send back the same response that I pulled out of wireshark, and updating my /etc/hosts file to point to localhost instead of the openevse.  If I set Content Length to 0 and removed the "Not Modified" line, Safari no longer gave the parse error.

It looks like the 7.x Mongoose doesn't have this issue, but it's kind of hard to say for sure since it looks like a pretty substantial rewrite.  They also seem to do a better job of sending back more of the response headers on 304, like the etag header etc.

This Mac issue was mentioned in https://github.com/OpenEVSE/openevse_esp32_firmware/issues/930  but I think these might be separate, unrelated issues at this point.

nathanejohnson, 2025-04-28:
it does look like with the latest Mongoose 304 now has an empty body:

https://github.com/cesanta/mongoose/blob/7.17/mongoose.c#L1977

nathanejohnson, 2025-04-28:
Also, for what it's worth, it looks like 6.18 mongoose no longer sends a body:

https://github.com/cesanta/mongoose/blob/6.18/mongoose.c#L8372

nathanejohnson, 2025-04-30:
So I updated my test program to be a little more useful, it actually proxies connections so that the websocket connection works but I can play with responses coming back.  It looks like Safari completely ignores Content-Length headers on 304, provided the body is not there.  Chrome is the same.

jeremypoulter, 2025-04-30:
Have a PR (#964) with version 6.18 of Mongoose

nathanejohnson, 2025-04-30:
Very nice! Would you like me to try that branch locally? I've never built this firmware but I could probably figure it out from the GitHub workflows

jeremypoulter, 2025-04-30:
You can download test binaries from the Workflow runs https://github.com/OpenEVSE/openevse_esp32_firmware/actions/runs/14742273648

nathanejohnson, 2025-04-30:
So unfortunately it doesn't seem to have changed the behavior.  It's still sending a body with the literal text "Not Modified".  

[image]

[image]

nathanejohnson, 2025-04-30:
This sends the 304 here:

https://github.com/jeremypoulter/ArduinoMongoose/blob/v0.0.21/src/MongooseHttpServer.cpp#L432

and this is what writes the body:

https://github.com/jeremypoulter/ArduinoMongoose/blob/v0.0.21/src/MongooseHttpServer.cpp#L432

nathanejohnson, 2025-04-30:
I'm running a local build with this small tweak:

https://github.com/jeremypoulter/ArduinoMongoose/pull/40

It's finally working on Safari with 304 responses.

jeremypoulter, 2025-05-01:
Thanks I will get that merged and a new version out

nathanejohnson, 2025-05-01:
I commented in the PR, but I just wanted to chime in here to say I'm running the latest build and the issue seems to be fixed!  Thanks again!

PR #964 (2025-05-27): Use latest (6.x) version of the ArduinoMongoose to resolver issues with Safari
Should fix #963