OpenEVSE/openevse_esp32_firmware #969: MQTT doesn't reconnect after network outage, reboot required

lutorm, 2025-05-09:
I've noticed that if the network between the openevse and the mqtt broker goes down, mqtt stops working and does not reconnect even after the network is back. Cycling mqtt to disabled and back to enabled doesn't help, but a reboot of the wifi does.

With shaper enabled the evse stops working without mqtt, so this means car won't charge until you discover this and reboot.

KipK, 2025-05-15:
Hi

I got this behavior too, it doesn't happend too frequently, probably because my evse have a strong wifi signal here. 

After reviewed a bit the code I think the  bug most likely stems from the mqtt_connect() function in mqtt.cpp not correctly resetting the static bool connecting; flag to false if the mqttclient.connect() call fails to even initiate a connection (i.e., if it returns an immediate error/false). This leaves connecting stuck as true, preventing all future connection attempts until a reboot.

To summerise my theory:

***1- connecting Flag Mismanagement:***

- In mqtt_connect(), the connecting flag is set to true at the very beginning.
- If the mqttclient.connect(...) call itself fails to initiate the connection (e.g., returns false immediately perhaps due to an internal Mongoose state after network flaps), the execution path might not reach any of the callbacks (onSuccess, onError, onClose) that are responsible for resetting connecting to false.
- If mqttclient.connect(...) returns false (or equivalent failure indication) and connecting is not explicitly reset to false in that failure path within mqtt_connect(), then connecting remains true forever.

***2-Consequence of Stuck connecting Flag:***

- Once connecting is stuck at true, any subsequent calls to mqtt_connect() (either from mqtt_loop()'s retry logic or from mqtt_restart()) will hit the if(connecting) { return false; } guard at the top and do nothing further.
- The system will no longer attempt to actually call mqttclient.connect(...).

***3- why mqtt_restart() fails to fx it:

- The mqtt_restart() mechanism calls mqttclient.disconnect()  and then schedules an immediate reconnection attempt by setting nextMqttReconnectAttempt = 0.
- However, when mqtt_loop() subsequently calls mqtt_connect(), if the connecting flag is already (and incorrectly) true, mqtt_connect() bails out immediately. The disconnect call might successfully trigger onClose and set connecting = false, but if the next attempt's mqttclient.connect() call fails without a callback and leaves connecting true, the cycle of being stuck begins.

@jeremypoulter what do you think of this ?

KipK, 2025-05-15:
I have started to refactor mqtt as a microtask, and probably fix this issue here: https://github.com/KipK/ESP32_WiFi_V4.x/commit/0ced566378d92e6cb9f0a767e7e4ee9e3b5d37f5

It compiles ok but I haven't tested it yet; so consider it as unsafe. I won't be able to test before newt week I think.

KipK, 2025-05-16:
Could test it finally. Seems to works pretty good.

I've compiled test builds for wifi_v1 and tft_v1 here:
https://drive.google.com/drive/folders/1FjYBbFW2SXE_5jxBuc-knzxWo3vFwJwK?usp=sharing

@lutorm , can you test this and tells me if it fix the issue?

jeremypoulter, 2025-05-16:
Thanks @KipK I think there is a lot of legacy code in there so a refactor/rewrite is probably in order. Will give your changes a test. It would be great if you could create a PR, if you don't think they are ready, a draft PR is fine.

KipK, 2025-05-16:
> Also be aware of [#964](https://github.com/OpenEVSE/openevse_esp32_firmware/pull/964) may be worth testing if that has an impact on MQTT

It is already merged on master ain't it? 
I used latest master for this change so it seems everything looks ok for what I could test. 

I'll submit a draft PR, I'd like to remove all the versions check on mqtt side now I've implemented the notify_xxx_changed() method.

lutorm, 2025-05-16:
> Could test it finally. Seems to works pretty good.
> 
> I've compiled test builds for wifi_v1 and tft_v1 here: https://drive.google.com/drive/folders/1FjYBbFW2SXE_5jxBuc-knzxWo3vFwJwK?usp=sharing
> 
> [@lutorm](https://github.com/lutorm) , can you test this and tells me if it fix the issue?

I can test, but I need a build for tft_v2 with European type 2 cable current limit. I think I can build myself though, I just have to remind myself how to do it and upload.

KipK, 2025-05-16:
@lutorm, there's no tft_v2 on the project, nor special build for europe, ( this is only for the openevse module , not for the wifi module  ).
There's only tft_v1, I've made a build for it. 
use esp32-wifi_tft_v1-mqttrefactor.bin from my drive.

lutorm, 2025-05-16:
Hmm I updated the wifi to the esp32-wifi_tft_v1-mqttrefactor.bin and now the screen is black and it's not coming up on the network...

KipK, 2025-05-16:
> Hmm I updated the wifi to the esp32-wifi_tft_v1-mqttrefactor.bin and now the screen is black and it's not coming up on the network...

Mmhh. That's weird. I can only test it on openevse_v1 module, I don't have the new TFT here, but I see no reasons it wouldn't work. Does current dev master works? 
You can reflash it using serial with a usb-ttl adapter

lutorm, 2025-05-18:
> That's weird. I can only test it on openevse_v1 module, I don't have the new TFT screen here, but I see no reasons it wouldn't work. Does current dev master works? You can reflash it using serial with a usb-ttl adapter

Ok, I see the wifi board needs 5v, are the serial pins 5v-tolerant as well or only 3.3v?

KipK, 2025-05-18:
Yes they are. There's supposed to be connected to openevse atmega module which runs at 5v.

lutorm, 2025-05-18:
> Yes they are. There's supposed to be connected to openevse atmega module which runs at 5v.

Ok great. I used a 3.3V adapter just to be sure and got it back up on master with the gui_v2 resources. I can try building the PR tomorrow.

KipK, 2025-05-19:
@lutorm, If you can catch up the console output from the serial port when it's not working it would help :) Not sure if you need to take the Serial or Serial2 for this.

As it doesn't ghet back to network  I think it just doesn't boot so Serial should be the one. 

Have you flashed already the correct partionning for the tft ? I've heard first batch where sent with the wifi_v1 partition ( min_spiffs.csv )  instead of openevse_16mb.csv

lutorm, 2025-05-19:
> [@lutorm](https://github.com/lutorm), If you can catch up the console output from the serial port when it's not working it would help :) Not sure if you need to take the Serial or Serial2 for this.
> 
> As it doesn't get back to network I think it just doesn't boot so Serial should be the one.
> 
> Have you flashed already the correct partionning for the tft board ? I've heard first batch where sent with the wifi_v1 partition ( min_spiffs.csv ) instead of openevse_16mb.csv

I don't remember, is there something special that needs to be done to change the partitioning? I don't think I've done anything beyond just updating the firmware. And I'm pretty sure I've done it OTA with my local changes and that worked, so I dunno what the problem would be. I can try to grab a serial dump, but I don't have a lot of time to experiment right now so it might be a while.

jeremypoulter, 2025-05-19:
Will try to check tonight

monfiston, 2025-06-08:
Hello,
Thank you for trying to fix this problem, as it's been around for a few years.
I just updated the firmware on my two OpenEVSEs. I'll let you know if I have any more problems.
According to the tests I ran this morning, your fix is ​​working.
I hope you'll include it in the official firmware very soon.

KipK, 2025-06-08:
@monfiston, good to know.  Do you have one with TFT screen? Or booth are v1 esp wifi modules?

commit 0ced5663 (2025-09-18): Refactored mqtt feature.
- made it a microTask
- Mongoose MQTT client callbacks are set up to call methods of the Mqtt instance (using lambdas to capture this).
- Error Handling in attemptConnection: The _mqttclient.connect() now has a non-blocking callback for success. If connect() itself returns false, it means the connection attempt couldn't even be initiated (e.g., invalid parameters before network activity). This case is now handled by calling onMqttDisconnect. ( should fix #969 )
-  checkAndPublishUpdates() method consolidates the logic for periodically checking if any state (claims, con [... cut]
closed 2025-06-12 by jeremypoulter