OpenEVSE/openevse_esp32_firmware #981: TFT display sleep/off

Knowsle, 2025-07-01:
Reading the latest update the display should turn off if the device is in sleep mode after 600 seconds, default build value. However i don't find this to be working. My display stays on no matter what mode it's in.

mooninite, 2025-07-17:
@kimble4 any thoughts?

DBNet-OOB, 2025-08-28:
@jeremypoulter - Just wanted to say THANK YOU for looking into this.  At first I had thought that it was due to my one of my openEVSE's now using SHAPER and it was for some reason keeping the screen on all time based on the updates from SHAPER. 

Then I installed the current Firmware on the other unit that was using the dev code as mentioned above, that does not use SHAPER, and it ended up having the same issue of the display never turning off.  

It was really nice to have the display not on all the time when not plugged in or after charging when it reached it's level. 

But just wanted to say thank you again for looking into it and looking forward to code I can load to try it.

David

jeremypoulter, 2025-08-28:
No problem, the LCD code was refactored in the last release so it is highly likely that this is a result of that, hopefully an easy fix

DBNet-OOB, 2025-08-28:
@jeremypoulter  ... Just a note based on ...."was refactored in the last release"... I also tried going to 5.1.3, just for kicks, and the display still did not turn off. (Note, I updated a unit that currently has no vehicle connected to it.)   I did not go back further than that.  

As mentioned, the last firmware I had running on that unit before I ran into this was "master_dfc1e1f7" and it did turn off the display.  From that verion I then went to 5.1.4 and then down to 5.1.3 both with the issue.   

Mentioning this just in case it may help.   Thank you for all you do!!!

DBNet-OOB, 2025-09-18:
@jeremypoulter - Just wanted to check in on this.  As soon as you may have a fix in a dev, I am willing to give it a try.

Thank you.

David

DBNet-OOB, 2025-09-19:
[image]

Well I think I found it and loaded it.  Seem to be the right thing. (It cam down in a zip file and I found a firmware.bin inside that along with 2 other files.)  I was not able to find it with your link, but somehow found it in any case.  I will report back as soon as I see it working. :)  (My wife is out with the Rivian right now.) 

Thank you

jeremypoulter, 2025-09-19:
Note that the screen will remain on while charging at greater than 0.1 amps.

DBNet-OOB, 2025-09-20:
@jeremypoulter - Update...All seems to be working well on two OpenEVSE TFT units.  The display is acting as expected and turning off and on as one would expect. 

Looking forward to it making it into the the general release.

Thank you!!!

David

PR #997 (2025-09-25): Fix for LCD timeout
Plus a bi of tidyup of some bits missed during the refactor

Fixes #981
commit 06da17fe (2025-09-25): Fix for LCD timeout

Plus a bi of tidyup of some bits missed during the refactor

Fixes #981
closed 2025-09-25 by jeremypoulter