OpenEVSE/openevse_esp32_firmware #985: WIFI and MQTT Disconnect

Cataract1200, 2025-07-25:
This has been an ongoing issue over the years where if something disrupts the wifi connection MQTT will disconnect.  However recently with my upgrade to the TFT/WIFI screen I've noticed that the wifi connection will also not recover.  I believe this is a quirk within the ESP32 that is known.  At least on the MQTT side this is a problem if one is using Shaper due to the fact that if Shaper does not receive an update it will disable charging.  Loss of WIFI has the same affect.

I was wondering if maybe a simple fix would be to include a watchdog timers, maybe 1 looking at Shaper to ensure an update occurs and if it doesn't reset the MQTT connection or just reset the device (device reset always fixes it) and another looking at the WIFI connection to ensure it's there and if not, reboot.

KipK, 2025-07-25:
Try the current dev build, mqtt disconnection should be solved.

jeremypoulter, 2025-07-25:
The latest v5.1.4 also includes the MQTT fix

Cataract1200, 2025-07-25:
Great to hear that the latest 5.1.4 includes the MQTT fix, I do have that installed so I will keep an eye on it.  One other piece if the wifi reconnect.  Not sure if there is a fix in the works at this point or maybe we could include a watchdog timer and reset for that.

th-in-gs, 2026-03-24:
This happens to me too with 5.1.5 - car didn’t charge yesterday after a WiFi outage :-(. Had to restart the firmware to get it to reconnect to MQTT. Happy to provide logs if that’s something I can do.