OpenEVSE/openevse_esp32_firmware #1003: Time isn't updating via NTP on reboot when that is the default and config is setup

dlarue, 2025-09-29:
I installed OpenEVSE JuiceBox v2 edition last week and had a brownout event which caused OpenEVSE to loose its network/WiFi connection so I had to reboot by cycling power to the OpenEVSE circuit. What I noticed once it connected to the WiFi network was that the time was incorrect. Going into the Time configuration page it showed it was set to NTP and the server and TZ were both still set from my initial setup.
15 minutes later it was still unset. Firmware is v5.1.4

[image]
[image]

emaste, 2026-06-04:
I just installed it (version reports 1087/merge_40ed347f), after a power cycle it still came up as 1969 and after a few minutes has not synced.

Legot, 2026-06-07:
> [@Legot](https://github.com/Legot) Many routers cache pool.ntp.org. Are you passing DNS to your OpenEVSE station via DHCP? It seems all the addresses that would require DNS resolution are not working.

Very sure. This happens with DNS caching completely off, as well as if I force one of the other addresses into the Unbound cache. Critically, the fact that it works when using ip addresses that are part of pool.ntp.org, which does not interact with DNS at all, is peculiar.

I also notice that addresses that are not pool.ntp.org don't generate DNS traffic on port 53.